mainsail-crew/MainsailOS #3: Update Wizard

RaymondHimle, 2020-09-03:
Implement a web-based update (klipper, moonraker, mainsail) and firmware configuration (make menuconfig) tool

KwadFan, 2020-11-25:
Hi, if I allowed to tell you my opinion on this, the following I would say:

Don't make this system dependent!
I think we use the posible best solution for this.
The moonraker API. Let moonraker do this Job to let the freedom to choose your Frontend of choice. So, Mainsail, Fluidd or Octoprint what ever you want. Since the only real Alternative to Mainsail for me is Fluidd, what is a Fork of Mainsail, you would loose this "band of brotherhood" and probably loose compatibilty.

So, if you ask me let moonraker serve this API.
Regards

raymondh2, 2020-11-25:
Fluidd is not a fork of Mainsail. FluiddPi is a fork of MainsailOS. Moonraker will be the API for powering updates for at least Klipper and itself as of right now.

KwadFan, 2020-11-25:
Sorry for misspelling the names, I meant *PI and OS not the frontends :)
So far...

mdframe, 2020-11-28:
Do we have a process for upgrading from .9 version of mainsailos to the newly released 1.0?

Thx

mdframe, 2020-12-04:
I was able to utilize the same mainsail install instructions for an in-place upgrade and it seemed to work.

wget -q -O mainsail.zip https://github.com/meteyou/mainsail/releases/latest/download/mainsail.zip && unzip mainsail.zip && rm mainsail.zip

Thx

closed 2020-12-27 by raymondh2