mainsail-crew/MainsailOS #12: sudo without passwords, even in ssh

KwadFan, 2020-11-25:
Since I did some research in a earlier Version, where i couldnt do HOSTREBOOT or HOSTSHUTDOWN from Web Interface, i discovered that the Module "password-for-sudo" was the enemy I am fighting with.

After I dealed with this and did a git fetch/checkout, you removed this Module.
Please dont do that! Call me a Paranoid, but what if some Blackhat-ed guy could get accsess to my Printer?
He can open Hells gates on Pi, remember its base is a Debian, with a few commands, you could turn your Printer in a Kali Linux HaxX0oR Box and now your really screwed. Besides what can someone do, who has skills to do something like that and understands Klipper fully?
Burn down my House? Not really good.

So, let me work further on this and bring back "password-for-sudo module". running Klipper & Co. as unprivileged user and running nginx as reverse-proxy has real "Firewall" quality und NOPASSWD for sudo kills all that entirely. 

My Solution till now was adding pi to sudoers file with permits using reboot/shutdown without password. I am working on a better solution so that a non privileged user gets sudo rights to do that.

This involves more things, in short:
Creating a User (mnrkrsudo)
making a config file in /etc/sudoers.d (020-elevate-mnrkr)
checking this file (see above)
modifying the lines in moonraker to use this user with sudo right on
(Totally py noob, so takes time)
fine choosen commands (sudo --user mnrkrsudo /sbin/systemctl reboot, for example).

Let it get Weekend and I bath in Coffee with a drinking straw and a Laptop....

raymondh2, 2020-11-25:
This solution would work if not for the fact that creating another user adds more complexity to the whole design. As moonraker still needs to access files owned by another user. We could simply just make those owned by the new user but that would also require the uses to use the sudo command to edit it via cli.

closed 2020-11-25 by raymondh2

KwadFan, 2020-11-25:
The Solution  I am working on wouldnt effect moonraker it self or the user pi/whatever installed moonraker. It effects only the calls from within *.py they are till now "sudo shutdon now". I run my solution manually editet. I have only to back that into an Image for testing out of the box. mnrkrsudo is in this case a systemuser without login and passwort. Same thing what nginx do when not running as root. I'll get it done. Then take a look please :)