mainsail-crew/MainsailOS #22: Background process regularly leads to Klipper MCU shutdown

Kuenstler79, 2021-02-27:
Hello,

I have Mainsail OS freshly installed on a Raspberry Pi 3 B+ (and nothing else) and then via "kiauh" Klipper and Moonraker.
Unfortunately it happens very often that some process in the background takes a lot of processor load and forces Klipper into a mcu shutdown. See also my error description here: [https://github.com/KevinOConnor/klipper/issues/3987](url) 
How can I find the resource-hogging process? Or is there a way to specifically slow down other processes besides Klipper?

Thanks!

Peter

meteyou, 2021-02-27:
Klipper and Moonraker are preinstalled in mainsailOS. what was the reason for installing it via KIAUH?

Kuenstler79, 2021-02-27:
Because I can easily install other components via "kiauh" and also flash the printer mainboard.
Is it then a problem to use "kiauh" under Mainsail OS? Are then components duplicated and active?

meteyou, 2021-02-28:
kiauh should work fine on mainsailOS, but only klipper & moonraker should be preinstalled. i dont know what kiauh do, when you click install and it is already installed.

Kuenstler79, 2021-02-28:
I'm not quite sure now if kiauh didn't even show what was already installed at the beginning.
I have then still installed Fluidd and wanted to test soon also times the KlipperSreen.
A webcam I still have in operation (connected via the Raspberry Pi Camera connector). Are there different connection methods or drivers here?
Can I find a log file somewhere, in which the running processes are recorded with the CPU load?
I am still quite a Linux noob.

Kuenstler79, 2021-03-03:
One idea could be to integrate a switch in the user interface to turn off the webcam for particularly important prints. Or perhaps show a task manager that you can still counteract and kill an unimportant task.

Kuenstler79, 2021-03-05:
The shutdown issue has been occurring more and more frequently over the past few days. Unfortunately, with "htop" I could not find out exactly which task grabbed all the processor load.
In any case, "mjpeg_streamer" and "nginx" were always on top.

I just switched to fluiddPi and fluidd. And no more problems since then. I don't know if it's the operating system in general or just the fact that I set up the system again.

The high priority processes are now always "klippy" or "moonraker". "nginx" and "mjpeg_streamer" are of course still active, but never on the top places to see.

Ticket can be closed from my point of view.

closed 2021-03-05 by Kuenstler79