mainsail-crew/MainsailOS #33: FluiddpiOS Feature request

TronskiFPV, 2021-04-23:
How about a default printer.cfg in fluiddpi? That way I can edit it with the tools already in fluidd. Small concern of course since fluiddpi does 99% of the work the user, but on first install I assumed klipper wasn't pre-installed correctly because that file was missing. Users upgrading from a Octoprint/klipper install (wild assumption that's the majority of fluidd users) can be up and running with a ctrl-c/ctrl-v. 

Something simple so there is no confusion and no accidental movements of the printer. links are probably not a great idea since install instructions may not be stable and third party guides end up what brings things together for lots of people. 

Above all, thanks for Fluiddpi. That was painless and I won't use anything else now on my printers. 

printer.cfg
This file needs to be edited for your specific printer, peripherals and taste. If you are unsure of what goes here or are receiving a error, follow the klipper instructions to flash your MCU (printer control board) and #to edit this file allowing Klipper to connect to your MCU

raymondh2, 2021-04-23:
This is not the FluiddPi repo.

closed 2021-04-23 by raymondh2

TronskiFPV, 2021-04-23:
opps assumed it was the same dev on both.