mainsail-crew/MainsailOS #41: Webcam FPS limited in Mainsail

johnderuntz, 2021-06-08:
The webcam view in the mainsail program seems to be fps limited in some way. If you set your webcam config to 1280x720 30fps like I do, the webcam is low fps inside of mainsail, but runs a lot smoother if you open a new tab and use <URL>:8080/?action=stream.

KenadyDwag44 on Discord

raymondh2, 2021-06-10:
Likely a Nginx issue. Will need to look into it more

meteyou, 2021-06-10:
pls dont use the "mjpegstreamder-adaptive" mode with high FPS. this mode only pulls single shots and dont grab the livestream.

KwadFan, 2022-04-10:
I think we could close this.
We added already a fix to the nginx conf to solve Issue due proxying and it seems to be working fine.

See https://github.com/mainsail-crew/MainsailOS/commit/1dd32e8083defea09998c17a1b4d01c2ad83aad9 for details.

Regards

closed 2022-04-10 by meteyou

ExploiTR, 2024-03-19:
[image]

I think the issue still persists, the left cam is running at 60fps if I use :8080/?action=stream directly and the right one at 30.

But very laggy in mainsail UI.

mryel00, 2024-03-19:
@ExploiTR First of all this is no MainsailOS issue. Everything that can be done within MainsailOS directly has already been done. Furthermore meteyou already wrote that using the `Adaptive MJPEG-Streamer` service, that you are most likely using inside Mainsail, will result in lower fps. Have you already tried to use the `MJPEG-Streamer` service?
I tried it myself right now. I get 1920x1080@15fps with `Adaptive MJPEG-Streamer` and1920x1080@50fps with `MJPEG-Streamer`. `MJPEG-Streamer` can be compared to the url directly in your browser.
I don't know why I don't get the full 60fps, as it seems capped at 50 for some reason, but that's a different story. I highly doubt that you can see any real difference between 50 and 60 fps anyway.
I will still investigate on the 50fps cap, but imho no one really needs above 15fps, and above 30fps you most likely won't see any major visual difference.

TLDR: Don't use `Adaptive MJPEG-Streamer`, instead use `MJPEG-Streamer`

ExploiTR, 2024-03-24:
@mryel00 Sorry for the late reply, I'm using `MJPEG-Streamer` only. Also I noticed something else - that If I only stream one camera at a time in MainSail UI, I'm getting 60fps and goes for both cameras. But as soon as I try to view both it becomes laggy.

I monitored RAM/CPU of PI(3) usage while both cams are open but looks good.

Also I should mention that one of them is a esp32cam and other is connected to Pi.

mryel00, 2024-03-29:
@ExploiTR I tested it again with 2 cams `2560x1440@30fps` and `1920x1080@25fps`. It's not a problem for my Pi3B+. Everything higher than 30fps is for me just luxurious and nothing I will even talk about, but I accidentally set the wrong parameters and I had shortly `1920x1080@50fps` and `1920x1080@25fps` running without any problems at all. Windows 11, Chrome, MJPEG-Streamer service and crowsnest/ustreamer as backend.
> I monitored RAM/CPU usage of PI(3) while both cams are open but looks good.

You should monitor your PC not the Pi, as Mainsail is running in your browser and not on your Pi.

Furthermore as this has nothing to do with MainsailOS, please go either to our [Discord](https://discord.gg/mainsail) or our [Discussions](https://github.com/orgs/mainsail-crew/discussions), if you want to discuss further on it.
As long as we cannot reproduce it we can't do anything about it and as my test shows, it isn't directly a problem of Mainsail.

kaanngunerr, 2025-02-21:
I was getting only 1 fps, and the video was freezing constantly. Adding `custom_flags: --format=YUYV `to my crowsnest.conf file solved the issue. Now, I'm getting 10-15 fps instead.

natdog1984, 2025-11-03:
> [image]
> 
> I think the issue still persists, the left cam is running at 60fps if I use :8080/?action=stream directly and the right one at 30.
> 
> But very laggy in mainsail UI.

I am still having this issue, and unable to resolve it.
Have tried switching the mode, changing the resolution, switching the USB Port, etc. 
I am using a HD Pro Webcam C920 on a RP 4B. 

I dont want 60fps, or even 30fps, but 15fps is reasonable I think. Most I can get is 11, and only for like 30 seconds after restarting the crowsnest. After that it goes to 10, until there is any movement on the video, then it drops to 0,1, 2, etc fps.

I switched from OctoPrint to Mainsail when moving to klipper firmware, and have really struggled with the Camera setup.
Note : OctoPi on this same hardware, was able to achieve 30fps. 

I am open to the idea that I have missed something, but I do not know what.

meteyou, 2025-11-03:
@natdog1984 MainsailOS is just a preinstalled rpi image. Nothing more. so this issue has nothing todo with MainsailOS. if you need help, pls use the Discord.