mainsail-crew/MainsailOS #47: Mainsailos looses wifi after a while

xp1ratex, 2021-07-22:
Mainsailos looses wifi after a while, top shows nothing out of the ordinary, free show nothing either.
If i reconnect the ip in unifi it comes back.
Rasberry pi 4 8Gb
klipper
v0.9.1-627-g121052ad
moonraker
v0.7.1-32-g42f61ce
mainsail
v1.6.0

[image]
[image]

any clues where to start looking? never had this problem before on octoprint.

KwadFan, 2021-08-06:
Please share the output of
`sudo iwconfig wlan0`

xp1ratex, 2021-08-08:
> Please share the output of
> `sudo iwconfig wlan0`

Here it is, got tired of trying to get it to work so i connected it with a ethernet cable instead, now it works flawless.
[image]

dstack, 2021-10-02:
I ran into a similar issue.  Specifically, Wifi-based networking would start, then seriously degrade, leading to timeouts, and then an eventual disconnect.

Steps to re-produce:
- Fresh install of MainsailOS
- SSID and password in wpa-supplicant, specifically 2.4Ghz network
- Raspberry Pi 4 B 2Gb, located approx. 30ft from AP.

I got about 2-3 minutes of solid < 10ms pings, then it would jump to 2000+ms and over 30 seconds, disconnect.  I tried this setup with and without KlipperScreen (which has a built-in wifi manager I thought could be interfering) and also with a DSI touchscreen (the touchscreen made it fail faster, but I was able to get this behavior consistently without the DSI screen).

Troubleshooting:
- I tried a variety of power setups, including connecting to a 5V 25W power supply.  Underpowering of course produced worse results, but standard and/or over-powering didn't change the behavior.
- I tried swapping out the boot disk with a standard RaspberryOS Lite, as well as an older install of RetroPi.  Neither of these installations could produce the same result, which is to say, they worked fine.
- I was able to repeat the result when swapping out the Pi.  I tried with 2 other RPi 4 B 2Gbs, an Rpi 400, and an RPi 4 B 8Gb.  All of them were able to reproduce the issue.
- I tried a variety of locations for this Pi (not a practical fix since the printer cannot move much from where it is), and only got some success by placing the Pi VERY close to the AP (< 6ft)

So, the major variables I altered in the troubleshooting were the location, the hardware, and the OS image, and the only noticeable result came from swapping out the OS.

The responsible response here would have been to switch back to using a Raspberry OS base, with Kiauh, and then putting the Mainsail stack on top, however ...

My Workaround:
- I updated the raspberry pi firmware, since this specific Pi 4 was one of the older ones in my collection.  This improved the situation a little, as I was able to get 5-7 minutes of solid connection on average.  This is normally not advisable, and you run the risk of bricking your Pi, so I will not provide the instructions on how to do this.
- Based on the fact that connecting the DSI screen made the issue worse, reading about an older issue concerning DSI screens and 2.4Ghz wifi specifically, and following a hunch, I changed this Pi to use the 5Ghz network instead.  None of my Pis do this, since I specifically relegate all the "IoT" / "Small-server" devices to the 2.4Ghz.  This provided a reasonably stable connection, even with the DSI screen attached.  I say reasonably stable, because I have noticed a periodic latency spike.  Generally, other devices on my network can ping this Pi with < 10ms latency, occasionally jumping to around 150ms, which I would consider normal, however, every 5 minutes or so, there's a huge latency spike resulting in a 3000+ms ping or two, which then goes back to <10ms.

Conclusions:
  It would be easy to dismiss this as part of the known issue with some DSI screens and 2.4Ghz wifi on Raspberry Pi 4s, except that it does not explain why RetroPi and base RaspberryOS Lite did not suffer from this issue.  It can't be location, it can't be hardware, and while it could be power, I'd find it hard to believe that all 5 of the power supplies I used were somehow not pushing 5.1V.

  Without digging in, I don't know if MainsailOS is doing anything special with wifi management, but this OS image was the only one that would reproduce the issue, every time, regardless of other variables.

meteyou, 2021-10-02:
thx for your report. the interesting thing is, that mainsailOS is a nearly pure raspberry pi os lite. we just disable the power management of the wifi chip, because this setting set the wifi chip sometimes in a sleep mode, when it is idle.

jottr, 2021-11-12:
@meteyou I couldn't find the place where you actually disable power management.  
Could you kindly point me to the relevant place in the source?

KwadFan, 2021-11-12:
> @meteyou I couldn't find the place where you actually disable power management. Could you kindly point me to the relevant place in the source code?

Hello again,
So here is your [pointer](https://github.com/guysoft/CustomPiOS/blob/612ed4b25ed742a7d3dd10146ea92a867a9e6659/src/modules/network/start_chroot_script#L57). This is implemented through the "network" Module of CustomPIOS, the framework that we use to modify the Lite Image.

Regards

AndreyShpilevoy, 2021-12-07:
Also, have the same issue with RPi4.

Markus-1984, 2021-12-12:
Same issue here, but with a RPi3B.
After a few minutes, mainsail can´t connect to moonraker.
After a while it´s connected again, but only for few minutes.

namelessFPV, 2022-01-01:
Same issue but with a RPi3B+ and Pi Zero 2W

Accidentally when the printer prints or is just in standby, Mainsail loses the connection to the wifi. The pi is still listed in the Fritzbox but you can no longer connect to the Printer

dz0ny, 2022-01-09:
> Same issue but with a RPi3B+ and Pi Zero 2W
> 
> Accidentally when the printer prints or is just in standby, Mainsail loses the connection to the wifi. The pi is still listed in the Fritzbox but you can no longer connect to the Printer

Exactly same issue and setup, can you check if you have huge spike in WIFI channel allocation when you see dropout 

[image]
under 
[image]

namelessFPV, 2022-01-09:
I will check it these days and let you know!

tommyl2210, 2022-01-12:
I am also loosing connection. Sometimes shortly after booting, sometimes when sending a gcode command during execution of another gcode command. e.g. change Z-offset during homeing. But also during printing... Print works normal but no access through wifi is possible.

dz0ny, 2022-01-12:
So enabling back the wifi power_save helps with the reconnection, before I had to restart by unplugging the power.

meteyou, 2022-01-13:
> This didnt happen with exact same hardware and octoprint.

this is really funny, because octoPI (the raspberry pi image of octoprint) and mainsailOS uses the exact same settings...

dz0ny, 2022-01-13:
I think this is something else, is something else masquerading as a wifi issue. I've tried RPI lite (official) and I get the same issue.

meteyou, 2022-01-13:
mainsailOS is based on a default rpi lite and then customizerPI just disable the power management of the wifi chip.

octoPI have only an additional service to ping the router or something else to have a traffic the full time.

dz0ny, 2022-01-13:
> octoPI have only an additional service to ping the router or something else to have a traffic the full time.

https://github.com/guysoft/OctoPi/blob/devel/src/modules/octopi/filesystem/root/etc/systemd/system/networkcheck.service
https://github.com/guysoft/OctoPi/blob/devel/src/modules/octopi/filesystem/root/etc/systemd/system/networkcheck.timer
https://gist.github.com/dz0ny/b90bcca9d60f6e8a2a33c55132d68c52#file-networkcheck

Lets see if it helps

Aussie84d, 2022-01-22:
I'm having the same issue. Could it be related to idle_timeout?

namelessFPV, 2022-01-30:
[image: image.png]
1min Scale 2.4Ghz

[image: image.png]
10min scale 2.4Ghz

[image: image.png]
1min Scale 5Ghz

[image: image.png]
10min Scale 5Ghz

da97ed, 2022-02-16:
> Exactly same issue and setup, can you check if you have huge spike in WIFI channel allocation when you see dropout
>
> [image] under [image]

I can confirm this. Got those spikes as well when loosing connection to mainsail. Unfortunately I`m loosing the connection quite often. I did a full re-install of MainsailOS and disconnected all my webcams. But the issue seems to consist. I am also unable to ping my raspberry during that off time. Terminal returns "host is down". Any help on this is much appreciated!! Im using Pi3B+ btw

meteyou, 2022-02-16:
@da97ed pls report this bug to raspberry pi os... we just use a generic raspberry pi os lite with add klipper, moonraker, nginx (for mainsail) and mjpegstreamer...

gurusonwheels, 2022-03-01:
i have same issue with pi4  driving me nuts .. have mainsailos also

meteyou, 2022-03-09:
@gurusonwheels @da97ed @Aussie84d @dz0ny @tommyl2210 @namelessFPV @Markus-1984 @xp1ratex 

to diagnose the problem we would need some data from you. can you please check if you have 2.4ghz and 5ghz active and if both networks have the same SSID name?

furthermore it would be good to know if "band steering" is active on your router.

dz0ny, 2022-03-09:
> to diagnose the problem we would need some data from you. can you please check if you have 2.4ghz and 5ghz active and if both networks have the same SSID name?
yes

> furthermore it would be good to know if "band steering" is active on your router.
yes, cannot be turned off

dz0ny, 2022-03-09:
I'am also getting much better stability after I've upgraded to latest kernel:
```
Linux prusa 5.15.18-v7+ #1520 SMP Fri Feb 4 11:58:51 GMT 2022 armv7l GNU/Linux
pi@prusa:~ $ sudo rpi-update next
```

meteyou, 2022-03-09:
@dz0ny thx four this important info! can you also try to disable "band steering" and test it for some days and if it dont work better, than try to rename one of the networks (2.4ghz or 5ghz) to have seperate names. maybe the rpi have issues, when the router change the band and lost the connection.

dz0ny, 2022-03-13:
Same outcome, different logs this time
```
[  291.819320] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x3e6 tid 0
[  380.948945] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x42 tid 0
[  381.029541] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x50 tid 0
[  381.296604] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x52 tid 0
[  381.406537] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x58 tid 0
[  381.476528] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x5c tid 0
[  397.426418] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x24d tid 0
[  619.636843] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x377 tid 0
[  619.916795] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x3ba tid 0
[  631.989933] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0xf27 tid 0
[  632.226819] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0xf32 tid 0
[ 1797.738846] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0xd98 tid 0
[ 1797.778791] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0xd9a tid 0
[ 1848.972071] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x90d tid 0
[ 2092.039516] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x178 tid 0
[ 2092.069574] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x17a tid 0
[ 2250.983654] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x65a tid 0
[ 2251.063732] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x662 tid 0
[ 2251.203634] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x665 tid 0
[ 2251.283651] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x669 tid 0
[ 2251.390614] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x66a tid 0
[ 2251.510542] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x66b tid 0
[ 2257.630640] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0xbd9 tid 0
[ 2257.680683] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0xbe1 tid 0
[ 2257.833668] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0xbe3 tid 0
[ 2257.910657] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0xbe9 tid 0
[ 2257.950799] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0xbeb tid 0
[ 2258.070622] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0xbf0 tid 0
[ 2258.260881] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0xbf1 tid 0
[ 2259.750630] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0xd72 tid 0
```

KwadFan, 2022-03-27:
Hello,
since there are many peoples, having the same Issue with connection lost, we decided to develop a solution.
It isn't the best solution but the most simpliest to avoid conflicts with changing system-depended firmware blobs or similar.

Please try out https://github.com/mainsail-crew/sonar, it is in WIP status, but we could archieve keeping an OrangePI with armbian online all the time during a period of one week, and they are even more picky in that topic.

Any suggestions to improve or change the service are welcome, please use Issue tracker of the project for that.

Regards

da97ed, 2022-04-05:
thank you! works for me

closed 2022-05-19 by meteyou

ccfman2004, 2023-05-03:
I know this is an old thread but I am having the exact same issue with WiFi on Mainsail 2.5.1 with everything up to date.

The only fix so far is to turn my printer off and the back on. I didn't always have this issue. It only appeared in April after an update.

PR #107 (2022-05-19): feat: add sonar by default to image
To prevent connection problems, it uses
https://github.com/mainsail-crew/sonar

This should fix #47
commit 33185db5 (2022-05-19): feat: add sonar by default to image (#107)

fixes #47