mainsail-crew/MainsailOS #56: "cannot upload file" among other issues

Nickbuddypal, 2021-09-27:
I've had mainsail installed for almost a year now and have had several issues down the track, mostly user error. Recently, I am no longer able to open, edit or upload any files through the mainsail UI. I always just get a "cannot upload file" for both .cfg files and .gcode files, or when I try and either open or edit a file I'm either prompted with a download popup in the bottom right hand corner that reads "Downloading printer.cfg NaN % @ /s" for a split second and then nothing. I've checked to see if my virtual_SD is set correctly but I believe it is. I am able to access and edit the files through winscp and through putty. Any ideas on what might need fixing?

Moonraker.log: https://pastebin.com/vsExueUm
Klippy.log: https://pastebin.com/e8pjhh3q

Nickbuddypal, 2021-09-27:
The issue also persists on a separate raspberry pi with mainsail installed, albeit with the same printer.cfg

Nickbuddypal, 2021-09-27:
I am also able to upload gcode files through prusaslicer but not within mainsail

Nickbuddypal, 2021-09-27:
I seem to have figured out the issue. It seems that mainsail does not function properly using Microsoft Edge (pls don't judge me), Its working with Safari and Chrome.

meteyou, 2021-11-13:
closed, because it is no mainsailOS issue

closed 2021-11-13 by meteyou