mainsail-crew/MainsailOS #57: no default 'printer.cfg'

Mangnoppa, 2021-10-20:
First of all: Thank you for the really great work! I just downloaded and installed MainsailOS und got past first boot. 

[The documentation](https://docs.mainsail.xyz/setup/mainsailos/first-boot#mainsailcfg) says to `[include mainsail.cfg]` into the _default_ `printer.cfg`. Unfortunately it seems there is no default printer.cfg. 

Some minor GUI-issue: After clicking the _Create File_-button a "window" opens, but it is not focused to start typing the new file name and hitting the return-key does not accept the entry.

zellneralex, 2021-10-20:
Hi, yes there is no default printer.cfg. Since we do not know for what printer you install it the printer.cfg would need to be an empty file and that would be as helpful as no file

Mangnoppa, 2021-10-20:
Thanks for replying so fast. Yes, you are absolutly right, having an empty file is as good as no file at all, but then the documentation is wrong. I think it would be good for beginners to put that into the documentation.