mainsail-crew/MainsailOS #62: Please provide release checksums

jottr, 2021-11-21:
Please provide `SHA256` release checksums for releases.  
I'd also like to suggest that you enforce [signed commits](https://docs.github.com/en/repositories/configuring-branches-and-merges-in-your-repository/defining-the-mergeability-of-pull-requests/about-protected-branches#require-signed-commits).

KwadFan, 2021-11-22:
Hello jottr,
to make it short, see [this](https://github.com/KwadFan/MainsailOS/blob/feature/sha256sum/.github/workflows/ReleaseRaspiOS.yml). I am facing a steep learning curve, I am by far no proffessional Developer. It is more a translation from my hobby to my Linux love.
But, as I said, last time, we hear our Users. So, be patient I am working on that.
Regards

KwadFan, 2021-11-22:
Hello again,
I think this will meet your requirements. 
https://github.com/mainsail-crew/MainsailOS/pull/63
Regards

jottr, 2021-11-23:
LGTM and thanks!

KwadFan, 2022-01-05:
Hello jottr, sorry I totally missed to answer.

Here is the solution: https://github.com/mainsail-crew/MainsailOS/commit/fecdf4032b288c9ec8a7b12d73009d162fd7ced6

Regards

closed 2022-01-05 by meteyou