mainsail-crew/MainsailOS #64: Cannot connect to Moonraker (192.178.200.10).

florin03, 2021-11-25:
on http://192.178.200.10:7125/printer/info i get :

"Traceback (most recent call last):\n\n  File \"/home/pi/moonraker-env/lib/python3.7/site-packages/tornado/web.py\", line 1681, in _execute\n    result = self.prepare()\n\n  File \"/home/pi/moonraker/moonraker/app.py\", line 386, in prepare\n    self.current_user = auth.check_authorized(self.request)\n\n  File \"/home/pi/moonraker/moonraker/components/authorization.py\", line 637, in check_authorized\n    raise HTTPError(401, \"Unauthorized\")\n\ntornado.web.HTTPError: HTTP 401: Unauthorized (Unauthorized)\n"

It is fresh install of mainsailos-rpi-lite-0.5.0, used different router with other ip conf and different raspberry.

meteyou, 2021-11-25:
Your IP address is not in the trusted range in your moonraker.conf. old check this config.

florin03, 2021-11-25:
can you give me an example on how to add it ?

raymondh2, 2021-11-25:
That's not a private IP address. Are you exposing moonraker to the internet?

florin03, 2021-11-26:
192.178.200.10 it my IP (changed from standard router setup), no am not exposing moonraker to the internet. changed router with one that has IP to range 192.188....... and no change.

VinnyCordeiro, 2021-11-26:
Usually the IP range for restricted/home networks is 192.1**6**8.x.x; 192.178.x.x is a regular, publicly available IP address, that's why people asked.

florin03, 2021-12-06:
managed to solve this with SSH to PI and using "sudo nano klipper_config/moonraker.conf" and under "trusted_clients:" and adding my IP address
in installation instructions it should be a note about this. 
thx for help[

closed 2021-12-06 by raymondh2