mainsail-crew/MainsailOS #82: Webcam isn't stable // uses a lot of CPU//crashes system

nickeeey, 2022-01-22:
Hi, i use a c270 webcam with Moonraker and a Raspberry Pi 3b+. The Webcam worked perfectly at octoprint, but won't work right with Moonraker. I see the picture/ Video after startup but when i switch 2-3 times to the dashboard it just loads. Also my cpu power is going up and i can't access files from the server. I tried different resoultions and framerates

meteyou, 2022-01-22:
I also use the c270 on 2 printers without issues.

nickeeey, 2022-01-22:
Can you upload your settings?
I dont know what to do. I also reinstalled the whole system without any positive results.
I think you use the normal mjpeg?

meteyou, 2022-01-22:
All default settings

KwadFan, 2022-01-22:
> I think you use the normal mjpeg?

Yes, MainsailOS uses mjpeg-streamer as default.
It is hard for me to reproduce due the fact I dont own Logitech Cams.

Are you able to describe what happend exactly when the system freezes?

Regards

nickeeey, 2022-01-23:
So i boot up the System and everything is working fine. Then my host is going for 30-50% on the circle at the machine tab. The Load is 1.3 or so and after a few minutes the Video is an endless loading circle. Also my prints stop because of the high load at the raspberry. When i print without Cam i get at max 20% i mean the 3b+ is more than enough for klipper

KwadFan, 2022-01-23:
So, we are hunting for that, we have a trace to the Problem. Please be patient. We are looking for a solution.
But, yes, basicly it is the same service for the webcam like in Octoprint ( besides Octos HLS Mode ).
The only thing what is different is the location of the configuration file.

Used Software behind that is jacksonliam's mjpg-streamer-experimental

So, give us some time to fix that.

Regards

nickeeey, 2022-01-23:
> So, we are hunting for that, we have a trace to the Problem. Please be patient. We are looking for a solution.
> But, yes, basicly it is the same service for the webcam like in Octoprint ( besides [...]

Thank you very much!
Just tell me if I can do anything for you. Maybe a video or sth?

nickeeey, 2022-01-24:
i opened cura and saw that there the webcam is still working. (Maybe from the Octoprint before). So there seems to be just a problem inside Mainsail os or the double stream is causing problems

meteyou, 2022-01-24:
it cannot be a problem inside of mainsailOS, because it is running in cura.

juslex, 2022-04-10:
Same problem here with c920

nickeeey, 2022-04-10:
I sold my printer and got a new one. On that it worked perfectly.
If you want I can send you my config

KwadFan, 2022-04-10:
Hey there,

Please keep in mind that, the higher the Resolution and the higher the framerate someone has to do the Job. In that case there is more or less no hardware accel in mjpg streamer. Please try out our future replacement.

https://github.com/mainsail-crew/crowsnest

The existing solution is quite old and the used backend isn't maintained anymore. So we decided to switch to a activly maintained solution called ustreamer as backend.

Regards

meteyou, 2022-05-19:
closed because of no response

closed 2022-05-19 by meteyou