mainsail-crew/MainsailOS #87: GCode tumbnail is not rendered

Aussie84d, 2022-01-23:
[image]

[image]

KwadFan, 2022-01-23:
Hello, thank you for that report.

But this isnt an MainsailOS Issue, this Bugtracker is only for Issues around mainsail (GUI) related Issue.
So, please report that here: https://github.com/mainsail-crew/mainsail/issues

Also, meteyou informed me that you should upgrade to Version 2.1.

Regards

closed 2022-01-24 by meteyou

Aussie84d, 2022-01-24:
Thank you. I will bring this there
[image]