mainsail-crew/MainsailOS #100: BUG ? MainsailOS 0.6.1 Not connecting to Moonraker

SISLANGER, 2022-03-15:
Running a fresh install from: MainsailOS-buster-lite-0.6.1 I cant get past Connection error as it is not connecting to Moonraker.

Cannot connect to Moonraker (mainsailos.local).

But if I install from 0.50 I have no issues.

meteyou, 2022-03-15:
Pls Upload your moonraker logfile

SISLANGER, 2022-03-15:
@meteyou 

[moonraker.log](https://github.com/mainsail-crew/MainsailOS/files/8253454/moonraker.log)

meteyou, 2022-03-15:
Pls Upload it again with your IP address. I have to check your settings with your requests

SISLANGER, 2022-03-15:
@meteyou all the ip addresses in the moonraker.log were the same, which is my current ip address. 
(Only a single replace all matching was used)

meteyou, 2022-03-15:
Pls tell me your address. I have to compare the trusted range with your address

meteyou, 2022-03-15:
ok. then you have to add `2a02:c7f:890:c100::/64` as trusted client in your moonraker.conf. mainsailOS > 0.6.x have ipv6 enabled and your client will connect via ipv6, but you router dont use a "standard" ipv6 network for your local address range.

meteyou, 2022-03-15:
do you restarted moonraker? if you have the same issue also after a moonraker restart, pls upload your COMPLETE logfile again.

as i told you above. v0.5.x have no ipv6 support and your client try to use IPv6...

SISLANGER, 2022-03-15:
yes, but I now done a full shutdown and it is now working. 

I know you said 0.5 doesnt use ipv6 but as I said, I updated it to 0.6.1 (on another SD card, same pi) and it worked without needing to add to trusted client. Does the update do this automatically? If so they would make sense.

meteyou, 2022-03-15:
there is no function to update mainsailOS settings. an up-to-date mainsailOS 0.5 is a mainsailOS 0.5. you have the same "old" nginx settings as before

closed 2022-03-15 by meteyou