mainsail-crew/MainsailOS #125: disable wifi powermanagement on rpi's without wifi

meteyou, 2022-06-17:
### What happened

you will have an error after booting a pi1 pi0 (without w) or pi2, when the /etc/rc.local will be executed.

```
[FAILED] Failed to start /etc/rc.local Compatability
```

### What did you expect to happen

no error with this file

### How to reproduce

just flash mainsailOS and use a rpi without wifi

KwadFan, 2022-06-25:
Fix depends on Pull Request in CustomPiOS 
https://github.com/guysoft/CustomPiOS/pull/168

If that https://github.com/mainsail-crew/MainsailOS/pull/127 should fix it.

KwadFan, 2022-07-19:
After a PR this could be closed due changes in custompios, see https://github.com/guysoft/CustomPiOS/commit/83d7fa9572eb3e385c9f2b57ff5ea4a68a60f892

closed 2022-07-19 by KwadFan