mainsail-crew/MainsailOS #134: Changing Wifi network not possible

MaikelKelder, 2022-08-08:
### What happened

After the initial setup and filled in Wifi SSID and password, logon is succesful after the installation.
Yet, changing to another WiFI SSID doesnt seem to work, not by file edit, not by klipperscreen.

By klipperscreen you see that he initiates Wifi, but it will never connect to new network.

### What did you expect to happen

Joining other WIfi network after entering SSID and password, manual file edit, OR by klipperscreen.

### How to reproduce

Change Wifi by file editting or klipperscreen.

### Additional information

I dont have expertise in Mainsail, yet found this to possible bug, I hope to help in this way and that my foundings are correct.

KwadFan, 2022-08-13:
Hey, we try to reproduce that this weekend.

Did you flash the Image with rpi-imager and set wifi credentials on setup?

MaikelKelder, 2022-08-13:
Imaged with balenaetcher.
Configured the WiFi with a file edit.
WiFi works after setup in pi.

KwadFan, 2022-08-13:
So, after flash you edited the mainsailos-wpa-supplicant.txt and it doesnt work.
What do you mean by "WiFi works after setup in pi." exactly?
I have to reproduce the steps as nearly I could, so I need an exact description what you have done.

What editor was used to edit the file?

Regards

MaikelKelder, 2022-08-13:
After creating the image on the sd, and editing the file as you describe,
you put it in the raspberry first time and WiFi works.

It’s when you try to change it after this, it doesn’t work, edited with
notepad++ and normal notepad (Windows).

Also tried to change it from klipperscreen, where you choose a WiFi from
the gui, that also fails.

Hope this makes sense to reproduce.

M.

KwadFan, 2022-08-13:
To my sadness we can confirm it, but I need  some time to investigate the root cause.

I will provide a solution as soon it is fixed for existing installations.

Sorry for the hazzle.

MaikelKelder, 2022-08-13:
No hassle at all, I appreciate the effort put into it to find it and trying
to solve it, respect all the way!

KwadFan, 2022-08-14:
I did some investigation,
In rpi-imager the softlink that points to the mainsailos-wpa-supplicant.txt, which is created by CustomPIOS ( the framework used to manipulate the base image) gets overwritten and does not work as intended anymore.

I have to do a test with balena also.

At this poitn I have no clue how to solve that easily but will provide a solution asap.

KwadFan, 2022-08-14:
Hey @MaikelKelder,

It seems thats not our fault. I found the following in the Raspberry PI OS docs.
`You will need to define a wpa_supplicant.conf file for your particular wireless network. Put this file onto the boot folder of the SD card. When the Raspberry Pi boots for the first time, it will copy that file into the correct location in the Linux root file system and use those settings to start up wireless networking.`

https://www.raspberrypi.com/documentation/computers/configuration.html#configuring-networking-2

I think they internally changed the behavior that leads to a behavior that CustomPIOS try to change with its "network module"
This changes from a real file "wpa_supplicant.conf" to a symlink to /boot/mainsail-wpa-supplicant.txt.

But this symlink gets overwritten if you try to change WiFi Settings with raspi-config and maybe ( I cant confirm that ) by KlipperScreen. Also it will be overwritten if you use rpi-imager to setup your WiFi and brake the existing symlink to the known txt file.

This is reproducable because normally it changes your clear text password in the txt file in /boot to an psk hash, but that dont happen either.

Could you maybe confirm that a wpa_supplicant.conf in boot do work?

Regards

MaikelKelder, 2022-08-14:
I can try this in a few days probably, dont have a printer to test it on
right now, but building a new Voron 2.4 at this moment.
If thats ok, I will give this method a try.

Great find, didnt know something changed in the PI OS.

Maikel.

KwadFan, 2022-08-23:
After some further digging it turns out that wpa_cli is the guilty one.
KlipperScreen uses wpa_cli to set WiFi settings, however wpa_cli overrides our softlink to /boot/mainsail-wpa-supplicant.txt.

I am working on a solution, but I think it will end up in making an example file and you have to put a wpa_supplicant.conf to /boot.

This is the mentioned way of Raspberry Foundation.

PR #142 (2022-09-01): fix: fixes error unusable wpa_supplicant.txt
This is more a workaround as a fix.
Raspberry Foundation changed handling of wpa_supplicant.conf.

Also KlipperScreen uses wpa_cli which overrides softlink to txt.

fixes #134
commit 53cb8195 (2022-09-01): fix: fixes error unusable wpa_supplicant.txt (#142)

This is more a workaround as a fix.
Raspberry Foundation changed handling of wpa_supplicant.conf.

Also KlipperScreen uses wpa_cli which overrides softlink to txt.

fixes #134
closed 2022-09-01 by KwadFan