mainsail-crew/MainsailOS #135: raspcam not work

swatkxs, 2022-08-10:
### What happened

Newest image (0.7.1), new install and raspcam not on by default (enable),the red led on camera not on, but work by disable pi camera(red led on).

issus tested on Pi zero w(first generation).

### What did you expect to happen

seems the switch of pi camera showed flipped signal.

### How to reproduce

maybe bugs happend on crowsnest

KwadFan, 2022-08-12:
I am sorry but I can't interprete what you trying to say. Cameras have nothing to do with MainsailOS.
This belongs to crowsnest and that enables raspicams by default, further more its backend ustreamer does not switch on light on the camera and its not intended to do that.

KwadFan, 2022-08-20:
Since there is now further Interaction, closed for now.

closed 2022-08-20 by KwadFan