mainsail-crew/MainsailOS #136: Bash scripts in ~/moonraker-env/bin hardcoded to assume username pi

mpfl, 2022-08-16:
### What happened

I installed MainsailOS 0.7.0 with the Raspberry Pi installer, changing the username from `pi` to `mpfl`.

When trying to install [moonraker-obico](https://github.com/TheSpaghettiDetective/moonraker-obico), I get the following error:
```
./install.sh: /home/mpfl/moonraker-env/bin/pip3: /home/pi/moonraker-env/bin/python: bad interpreter: No such file or directory
```

### What did you expect to happen

I expected moonraker-obico to install successfully.

### How to reproduce

1. Use Raspberry Pi Imager v1.7.2 to install MainsilOS 0.7.0 after setting username from `pi` to any other username.
2. Follow instructions to install [moonraker-obico](https://github.com/TheSpaghettiDetective/moonraker-obico)

### Additional information

It looks like some scripts in MainsailOS hardcode the python interpreter to be /home/pi/moonraker-env

~/moonraker-env/bin/pip
~/moonraker-env/bin/pip3
~/moonraker-env/bin/pip3.7

When I changed the interpreter to /home/mpfl it all worked okay.

meteyou, 2022-08-16:
mainsailOS has no influence on 3rd part installer. this issue needs to be fixed in obico installer. please open an issue in the moonraker-obico repo.

mpfl, 2022-08-16:
Sorry, @meteyou but the issue is with MainsailOS and not moonraker-obico. The [MainsailOS installation documentation](https://docs.mainsail.xyz/setup/mainsailos/pi-imager) states:

> Please don’t change the username!
> At this stage MainsailOS Setup relies on the user “pi”.
> We will change that in the future.

Sorry for not reading through the documentation before raising the issue, but the fact remains this is not an issue with moonraker-obico. If anything, the problem exists between keyboard and chair. My keyboard and my chair, that is.

hevilp, 2022-08-16:
The change of the username from pi to mfpl was successful? Did you install obico after you installed MainsailOS? You installed a 3rd party software and complaining about MainsailOS has an issue, while you installing the 3rd party software under the user mfpl?

th33xitus, 2022-08-16:
The obico installer has nothing to do with the issue itself as i was able to reproduce it just by flashing MainsailOS 0.7.0 with the Raspberry Pi Imager and change the username without running the obico installer at all.

It's just that the obico installer seems to have pointed out that there does exist a problem. Which is a fair to point out.

With the recent release of MainsailOS 0.7.0 we allow changing of the username and it should get handled respectively. The docs need to be updated in that regard.

KwadFan, 2022-08-17:
I am working on another solution to get that fixed.

I want to stick with the ability to rename the user, for security reasons mainly, but I think a custom Printer needs an custom user name also.

If someone wants to take a look what I am trying to do:

https://github.com/mainsail-crew/MainsailOS/tree/fix/rename

Regards Kwad

commit 170ca5ba (2022-08-19): fix: fix errors in moved venvs

This should fix #136

Due moving venv's with virtualenv,
some path aren't get changed.

Now pycache will be deleted and
shebang lines manipulated to match paths
PR #138 (2022-08-23): fix: fix errors in moved venvs
This should fix #136

Due moving venv's with virtualenv,
some path aren't get changed.

Now pycache will be deleted and
shebang lines manipulated to match paths
commit cb8a8220 (2022-08-23): fix: fix errors in moved venvs (#138)

This should fix #136

Due moving venv's with virtualenv,
some path aren't get changed.

Now pycache will be deleted and
shebang lines manipulated to match paths
closed 2022-08-23 by KwadFan