mainsail-crew/MainsailOS #151: BUG? Object shapes are not showed in exclude object box. Only names are listed

Markus-1984, 2022-09-25:
### What happened

Object shapes are not showed in exclude object box. Only names are listed.
It dosn´t matter if i preprocess the files via the windows preprocessor or the moonraker preprocessor.
In both ways it only shows the name of the object, but not the shape.

### What did you expect to happen

Object shape box should be displayed in exclude object box

### How to reproduce

Here is the gcode of my last try:

; object:{"name":"Shape-Box","id":"Shape-Box id:0 copy 0","object_center":[61.325362,118.423277,0.000000],"boundingbox_center":[61.325362,118.423277,0.300000],"boundingbox_size":[28.284271,28.284271,0.600000]}
; object:{"name":"main_body","id":"main_body.stl id:1 copy 0","object_center":[116.808822,120.964525,0.000000],"boundingbox_center":[116.808822,120.964525,12.351000],"boundingbox_size":[78.736343,78.829681,24.702000]}
; object:{"name":"motor_plate","id":"motor_plate.stl id:2 copy 0","object_center":[88.411327,61.471586,0.000000],"boundingbox_center":[88.411327,61.471586,4.100000],"boundingbox_size":[76.091904,64.733498,8.200001]}
; object:{"name":"sb_adxl_mount_generic_15","id":"sb_adxl_mount_generic_15.5mm_c_c.stl id:3 copy 0","object_center":[73.331750,96.470830,0.000000],"boundingbox_center":[73.331750,96.470830,4.850000],"boundingbox_size":[27.754076,30.352846,9.700001]}
; object:{"name":"sb_adxl_washer_x2","id":"sb_adxl_washer_x2.stl id:4 copy 0","object_center":[160.878818,123.291663,0.000000],"boundingbox_center":[160.878818,123.291663,1.500000],"boundingbox_size":[5.399971,5.399971,3.000000]}
; object:{"name":"sb_adxl_washer_x2","id":"sb_adxl_washer_x2.stl id:4 copy 1","object_center":[160.151203,115.927677,0.000000],"boundingbox_center":[160.151203,115.927677,1.500000],"boundingbox_size":[5.399971,5.399971,3.000000]}
; object:{"name":"stealthburner_printhead_dragon_front","id":"stealthburner_printhead_dragon_front.stl id:5 copy 0","object_center":[167.144094,70.578645,0.000000],"boundingbox_center":[167.144094,70.578645,13.700000],"boundingbox_size":[81.008781,81.079361,27.400000]}
; object:{"name":"stealthburner_printhead_dragon_rear_cw2","id":"stealthburner_printhead_dragon_rear_cw2.stl id:6 copy 0","object_center":[176.815541,149.928319,0.000000],"boundingbox_center":[176.815541,149.928319,13.127849],"boundingbox_size":[76.790024,67.774376,26.255697]}
; object:{"name":"probe_bracket","id":"probe_bracket.stl id:7 copy 0","object_center":[152.279002,179.396382,0.000000],"boundingbox_center":[152.279002,179.396382,2.000000],"boundingbox_size":[17.362354,17.362354,4.000000]}
; object:{"name":"x_frame_V2TR_MGN12_left","id":"x_frame_V2TR_MGN12_left.stl id:8 copy 0","object_center":[115.347817,186.611177,0.000000],"boundingbox_center":[115.347817,186.611177,10.799995],"boundingbox_size":[61.115700,68.567986,21.599991]}
; object:{"name":"x_frame_V2TR_MGN12_right","id":"x_frame_V2TR_MGN12_right.stl id:9 copy 0","object_center":[80.955571,173.437153,0.000000],"boundingbox_center":[80.955571,173.437153,10.800008],"boundingbox_size":[68.568008,61.115692,21.600016]}
; object:{"name":"chain_anchor_3hole","id":"chain_anchor_3hole.stl id:10 copy 0","object_center":[63.042432,137.992146,0.000000],"boundingbox_center":[63.042432,137.992146,10.930593],"boundingbox_size":[56.581371,32.599182,21.861185]}
; plater:{"center":[124.981150,125.000003,0.000000],"boundingbox_center":[124.981150,125.000003,13.700000],"boundingbox_size":[180.458807,191.790333,27.400000]}

; Pre-Processed for Cancel-Object support by preprocess_cancellation v(0, 3, 'a1')
; 12 known objects
EXCLUDE_OBJECT_DEFINE NAME=Shape_Box_id_0_copy_0 CENTER=61.326,118.423 POLYGON=[[61.325,104.611],[47.513,118.423],[61.325,132.235],[75.138,118.423],[61.325,104.611]]
EXCLUDE_OBJECT_DEFINE NAME=chain_anchor_3hole_stl_id_10_copy_0 CENTER=63.001,137.988 POLYGON=[[35.635,121.893],[34.952,122.575],[34.952,135.466],[45.905,146.42],[59.258,154.083],[78.368,151.002],[91.05,138.32],[88.774,135.854],[85.763,132.884],[84.126,131.395],[42.026,121.893],[35.635,121.893]]
EXCLUDE_OBJECT_DEFINE NAME=x_frame_V2TR_MGN12_right_stl_id_9_copy_0 CENTER=80.956,173.437 POLYGON=[[91.729,143.079],[90.966,143.131],[90.56,143.21],[90.103,143.343],[89.514,143.592],[89.178,143.779],[88.546,144.237],[74.298,158.466],[47.108,189.898],[46.943,190.133],[46.872,190.401],[46.896,190.677],[47.013,190.929],[47.108,191.042],[59.677,203.611],[59.868,203.744],[60.156,203.795],[74.341,200.828],[74.411,200.779],[113.794,164.225],[114.803,163.216],[114.968,162.982],[115.04,162.714],[115.015,162.437],[114.968,162.306],[114.803,162.072],[103.354,150.623],[102.533,149.826],[101.102,148.538],[99.457,147.203],[98.258,146.32],[96.706,145.269],[94.887,144.169],[93.933,143.644],[93.247,143.344],[92.492,143.149],[91.729,143.079]]
EXCLUDE_OBJECT_DEFINE NAME=x_frame_V2TR_MGN12_left_stl_id_8_copy_0 CENTER=115.348,186.611 POLYGON=[[132.311,152.527],[132.174,152.551],[131.923,152.669],[131.809,152.764],[100.376,179.954],[86.398,193.932],[86.139,194.212],[85.72,194.783],[85.349,195.506],[85.115,196.238],[85.042,196.622],[84.993,197.148],[84.993,197.514],[85.055,198.116],[85.263,198.927],[85.555,199.589],[86.631,201.482],[87.791,203.286],[89.017,204.986],[90.361,206.655],[91.817,208.274],[103.983,220.459],[104.217,220.623],[104.348,220.671],[104.624,220.695],[104.762,220.671],[105.013,220.554],[106.135,219.45],[142.739,179.997],[145.689,165.915],[145.706,165.725],[145.655,165.523],[145.521,165.332],[132.953,152.764],[132.719,152.599],[132.536,152.542],[132.311,152.527]]
EXCLUDE_OBJECT_DEFINE NAME=probe_bracket_stl_id_7_copy_0 CENTER=152.279,179.397 POLYGON=[[156.051,170.916],[155.71,170.937],[155.382,171.021],[155.075,171.166],[154.799,171.367],[144.331,181.829],[144.107,182.1],[143.993,182.292],[143.841,182.716],[143.798,183.165],[143.869,183.611],[143.944,183.82],[144.109,184.12],[144.331,184.388],[147.287,187.345],[147.558,187.569],[147.747,187.68],[148.068,187.807],[148.282,187.855],[148.626,187.877],[148.848,187.856],[149.281,187.73],[149.48,187.629],[149.759,187.426],[160.227,176.964],[160.451,176.693],[160.565,176.501],[160.653,176.297],[160.739,175.962],[160.753,175.515],[160.654,175.074],[160.566,174.87],[160.381,174.579],[157.184,171.367],[156.907,171.166],[156.601,171.021],[156.27,170.936],[156.051,170.916]]
EXCLUDE_OBJECT_DEFINE NAME=stealthburner_printhead_dragon_rear_cw2_stl_id_6_copy_0 CENTER=176.796,149.941 POLYGON=[[178.115,116.318],[147.295,139.829],[143.949,142.39],[143.624,142.689],[143.369,143.027],[138.765,152.601],[138.703,152.809],[138.688,153.037],[138.715,153.18],[141.526,157.856],[142.069,158.323],[143.572,159.382],[178.877,182.698],[181.582,183.564],[184.37,182.752],[213.529,151.732],[214.903,147.449],[186.276,118.822],[178.115,116.318]]
EXCLUDE_OBJECT_DEFINE NAME=sb_adxl_washer_x2_stl_id_4_copy_0 CENTER=160.879,123.291 POLYGON=[[160.9,120.792],[160.438,120.831],[160.009,120.948],[159.561,121.167],[159.149,121.487],[158.852,121.828],[158.602,122.259],[158.442,122.734],[158.38,123.238],[158.399,123.602],[158.529,124.145],[158.626,124.371],[159.101,125.001],[159.448,125.3],[159.8,125.545],[160.292,125.722],[160.822,125.791],[161.304,125.755],[161.769,125.628],[162.23,125.395],[162.597,125.108],[162.936,124.712],[163.151,124.333],[163.317,123.845],[163.378,123.352],[163.337,122.837],[163.215,122.402],[162.992,121.956],[162.681,121.559],[162.324,121.252],[161.88,121.001],[161.406,120.848],[160.9,120.792]]
EXCLUDE_OBJECT_DEFINE NAME=sb_adxl_washer_x2_stl_id_4_copy_1 CENTER=160.151,115.928 POLYGON=[[160.173,113.428],[159.711,113.467],[159.282,113.584],[158.834,113.803],[158.411,114.133],[158.121,114.469],[157.867,114.912],[157.714,115.373],[157.652,115.874],[157.671,116.238],[157.802,116.781],[157.898,117.007],[158.374,117.637],[158.721,117.936],[159.072,118.181],[159.565,118.358],[160.094,118.427],[160.576,118.391],[161.041,118.264],[161.45,118.064],[161.861,117.752],[162.191,117.373],[162.42,116.977],[162.581,116.516],[162.65,116.004],[162.619,115.528],[162.482,115.025],[162.255,114.577],[161.953,114.195],[161.597,113.888],[161.153,113.637],[160.678,113.484],[160.173,113.428]]
EXCLUDE_OBJECT_DEFINE NAME=main_body_stl_id_1_copy_0 CENTER=116.838,120.960 POLYGON=[[116.271,81.77],[115.2,82.391],[107.209,87.956],[99.414,95.751],[96.173,99.335],[82.508,115.17],[77.924,121.289],[77.721,121.596],[116.275,160.15],[119.301,159.208],[127.377,153.374],[143.943,139.158],[149.874,133.252],[155.956,122.917],[153.605,100.914],[136.636,83.947],[116.271,81.77]]
[... cut ...] 932,31.277],[115.911,30.812],[94.646,29.305]]
EXCLUDE_OBJECT_DEFINE NAME=stealthburner_printhead_dragon_front_stl_id_5_copy_0 CENTER=167.115,70.593 POLYGON=[[169.453,30.293],[166.333,31.55],[165.689,31.862],[131.394,56.413],[127.624,64.14],[127.057,65.328],[126.865,65.875],[134.243,78.145],[159.254,103.156],[159.613,103.515],[171.883,110.893],[172.43,110.701],[173.618,110.134],[181.344,106.365],[205.897,72.069],[207.366,68.487],[206.939,67.779],[179.334,39.892],[174.531,35.089],[169.453,30.293]]

meteyou, 2022-09-25:
Pls upload the gcode for testing

Markus-1984, 2022-09-26:
[2x_Voron_Design_Cube_v7.gcode.txt](https://github.com/mainsail-crew/MainsailOS/files/9649228/2x_Voron_Design_Cube_v7.gcode.txt)

Here is a sample gcode

Markus-1984, 2022-09-26:
The parts do not appear in the cancel object list in mainsail until the first layer of the part is being printed.

Markus-1984, 2022-09-27:
I've tried the gcode on another printer with mainsail v2.2.1 running and it works flawless. After I've clicked on the file and say print, the exclude object list shows the shape of the objects. 
Looks like a bug in the actual mainsail v2.3.0-beta1

Markus-1984, 2022-09-27:
Just updated to v2.3.1 and it's the same behavior.