mainsail-crew/MainsailOS #170: Logrotation config files might be writable by group

miklschmidt, 2022-11-12:
### What happened

So i don't know if this applies to you guys, but the logrotation configs supplied by the mainsail module ended up being 664 instead of 644 in the image, so logrotate would ignore those and the logs would just pile up.

### What did you expect to happen

Logrotation.d files should be chmodded to 644 (i'm doing this in the `start_chroot_script` now).

### How to reproduce

`ls -la /etc/logrotation.d`

### Additional information

This might very well be a RatOS only issue, but i just wanted to keep you guys posted in case it carried over from MainsailOS. Feel free to close!

miklschmidt, 2022-11-12:
See https://github.com/Rat-OS/RatOS/issues/39

KwadFan, 2022-11-16:
Hey @miklschmidt ,

Will do some tests at the weekend, may an issue due who Custompios copies files.
In the source the files are as expected 0644 ... :man_shrugging:

KwadFan, 2022-11-23:
Hey @miklschmidt ,

Made some tests. Short tests but they work. @zellneralex is doing a Longterm Test right now, will report results.
[image]
[image]

Regards Kwad

miklschmidt, 2022-11-23:
Well the permissions look fine, so the problem i had shouldn't be present at least :)

KwadFan, 2022-12-26:
@miklschmidt 

Could you reproduce that with 1.0.0 ?

KwadFan, 2023-03-12:
Closing issue due no further interaction and not reproducable since new release.

closed 2023-03-12 by KwadFan

miklschmidt, 2023-03-12:
> Closing issue due no further interaction and not reproducable since new release.

Sorry i forgot to get back to you on this, everything looks good on my end, so closing is appropriate 👍