mainsail-crew/MainsailOS #176: Error: 401 "Unauthorized" When atempting to download config files.

thetouchablegod, 2022-12-05:
### What happened

Trying to download configs from the web UI off of two machines running MainsailOS results in the following error:

`{"error": {"code": 401, "message": "Unauthorized", "traceback": "Traceback (most recent call last):\n\n  File \"/home/pi/moonraker-env/lib/python3.7/site-packages/tornado/web.py\", line 1690, in _execute\n    result = self.prepare()\n\n  File \"/home/pi/moonraker/moonraker/app.py\", line 539, in prepare\n    self.current_user = auth.check_authorized(self.request)\n\n  File \"/home/pi/moonraker/moonraker/components/authorization.py\", line 751, in check_authorized\n    raise HTTPError(401, \"Unauthorized\")\n\ntornado.web.HTTPError: HTTP 401: Unauthorized (Unauthorized)\n"}}`

Issue started after using the update manager to update all components to latest versions. Additionally, the built-in editor is unable to open any files in the /config directory.

### What did you expect to happen

Expected download of files.

### How to reproduce

In the machine tab, in the config directory, right click any file and select download. A new tab appears with the error.

### Additional information

I can grab logs if need be, but will have to ssh as downloading them from web UI makes the same error.

thetouchablegod, 2022-12-05:
[moonraker.log](https://github.com/mainsail-crew/MainsailOS/files/10155947/moonraker.log)
[klippy.log](https://github.com/mainsail-crew/MainsailOS/files/10155948/klippy.log)

KwadFan, 2022-12-08:
Hey,
this is not an bug of mainsailOS, this is a misconfiguration of your authorization in your moonraker.conf.

Also make sure you followed the steps as described in Issue https://github.com/arksine/moonraker/issues/516

This will hit you after a fresh flash of MainsailOS also.
Sorry for inconvinience, we are working on a new release.

Regards Kwad

closed 2022-12-08 by KwadFan

thetouchablegod, 2022-12-08:
I am aware this is intended behavior. I filed this problem under bugs just to make someone aware that something like this can happen in, case other users have similar issues.

I have never changed anything about moonraker.conf, and have been using both machines without issue for ~6 months. Suddenly after updating, I have intermittent problems.

thetouchablegod, 2022-12-08:
[server]
host: 0.0.0.0
port: 7125
# Verbose logging used for debugging . Default False.
# The maximum size allowed for a file upload (in MiB).  Default 1024 MiB
max_upload_size: 1024

[file_manager]
# post processing for object cancel. Not recommended for low resource SBCs such as a Pi Zero. Default False 
enable_object_processing: False

[authorization]
cors_domains:
    https://my.mainsail.xyz
    http://my.mainsail.xyz
    http://*.local
    http://*.lan
trusted_clients:
    10.0.0.0/8
    127.0.0.0/8
    169.254.0.0/16
    172.16.0.0/12
    192.168.0.0/16
    FE80::/10
    ::1/128

# enables partial support of Octoprint API
[octoprint_compat]

# enables moonraker to track and store print history.
[history]

# this enables moonraker's update manager
[update_manager]
refresh_interval: 168 

[update_manager mainsail]
type: web
repo: mainsail-crew/mainsail
path: ~/mainsail

thetouchablegod, 2022-12-08:
Here is my moonraker.conf. After further testing. I am unable to authenticate properly when I point my browser to my machine's hostname (i.e. http://tevo.local). I can connect to the web interface through hostname, and can control the machine just fine to run prints already uploaded or move axes etc. I cannot upload or download any files though. 

I have no issues when connecting via local IP. Uploading and downloading works as expected.

I know I'm an idiot, so maybe it's a DNS issue.   

```
[server]
host: 0.0.0.0
port: 7125
# Verbose logging used for debugging . Default False.
# The maximum size allowed for a file upload (in MiB).  Default 1024 MiB
max_upload_size: 1024

[file_manager]
# post processing for object cancel. Not recommended for low resource SBCs such as a Pi Zero. Default False 
enable_object_processing: False

[authorization]
cors_domains:
    https://my.mainsail.xyz
    http://my.mainsail.xyz
    http://*.local
    http://*.lan
trusted_clients:
    10.0.0.0/8
    127.0.0.0/8
    169.254.0.0/16
    172.16.0.0/12
    192.168.0.0/16
    FE80::/10
    ::1/128

# enables partial support of Octoprint API
[octoprint_compat]

# enables moonraker to track and store print history.
[history]

# this enables moonraker's update manager
[update_manager]
refresh_interval: 168 

[update_manager mainsail]
type: web
repo: mainsail-crew/mainsail
path: ~/mainsail
```

KwadFan, 2022-12-08:
So, I asked internally again.

This is an configuration error bekonging to IPv6, your network uses IPv6 and those have to be also in `trusted_clients`

In your case `2600:6c46:6200:ab1::/64` should do the Job.

For further help I highly encourage you to use, as mentioned, our Discord server. This allows others to help you and the team gets some free time to develop new things ;) 
Thanks in advance, over and out...