mainsail-crew/MainsailOS #193: Flashing Duet Wifi on recent MainsailOS (Bullseye) lets Duet disappear

Roycinger, 2023-01-04:
### What happened

I made a fresh install of MainsailOS via Raspberry Pi imager.
After flashing the MCU (Duet Wifi), the Duet disappeared.
It cannot be found by lsusb, nor by 'ls /dev/serial/by-id/*'.
Dmesg reports: 'unable to enumerate USB device'

While researching, I found this issue: https://github.com/Klipper3d/klipper/issues/4933
Conclusion is that a Duet flashed by Bullseye cant be discovered by the OS / Klipper anymore until it's flashed by Buster based Debian.

### What did you expect to happen

Successfully flashed and discoverable Duet Wifi

### How to reproduce

- Install MainsailOS (Bullseye based)
- flash Duet MCU

toskium, 2023-01-04:
Can confirm the issue. But it is not Mainsail OS specific, it also happens with vanilla Raspbian v11 (bullseye) and manual klipper installation.

meteyou, 2023-01-04:
This issue has to be fixed in Klipper. MainsailOS is just a Raspberry PI OS lite with Klipper preinstalled. Nothing more...

closed 2023-01-04 by meteyou

Roycinger, 2023-01-04:
> This issue has to be fixed in Klipper. MainsailOS is just a Raspberry PI OS lite with Klipper preinstalled. Nothing more...

That's what Toskium and I figured. We agreed that this is not a specific MainsailOS or Klipper issue, that's why I crossposted.
For anyone with the same / similar issue, follow along here:
https://klipper.discourse.group/t/lose-connection-with-duet-2-wifi-after-flashing/5721