mainsail-crew/MainsailOS #197: BTT TFT70 Touch Does Not Work With klipperscreen

Treintjes, 2023-02-10:
### What happened

Updated via clean install to 1.0 and 1.0.1 mainsailos and with both versions once klipperscreen is installed touch is inoperative tried modifying config, and via sudo neither lead to success of touch screen operation. moved back to mainsail 0.7.1 via clean install, installed klipperscreen and everything works fine... I will be posting this on klipperscreens git as well as im not sure where the problem lies.

### What did you expect to happen

normal operations of klipperscreen.

### How to reproduce

perform clean install of newest mainsail ver. 1.0.1 and then follow with clean install of klipperscreen and attemp to utilize touch on klipperscreen.

### Additional information

I suspect with moving from Buster to Bullseye there may have been a kernel/driver support for my screen dropped? I am using a BTT TFT70 v2.0 with dsi cable.

meteyou, 2023-02-10:
Mainsail OS has nothing implemented to support KlipperScreen per default. Mainsail OS is just based on a clean Raspberry Pi OS lite and with v1.0 we changed from buster to bullseye.

So please post this in the KlipperScreen Issue tracker.

closed 2023-02-10 by meteyou

zellneralex, 2023-02-10:
Read and try https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/#touch-not-working-on-debian-bullseye