mainsail-crew/MainsailOS #232: input shape

docemiel, 2023-06-12:
### What happened

orangepi 3lts
new install mainsailos! good!

the script off klipper for input shape

sudo apt update
sudo apt install python3-numpy python3-matplotlib libatlas-base-dev  -  good too

the script to numpy

~/klippy-env/bin/pip install -v numpy     -  do not work, do not install

### What did you expect to happen

the correct install

### How to reproduce

orangepi 3lts
new install mainsailos! good!

the script off klipper for input shape

sudo apt update
sudo apt install python3-numpy python3-matplotlib libatlas-base-dev  -  good too

the script to numpy

~/klippy-env/bin/pip install -v numpy     -  do not work, do not install

meteyou, 2023-06-13:
All you need for input shaper is preinstalled in MainsailOS and the pip install numpy is only necessary if you don't have a python-numpy package.

docemiel, 2023-06-13:
everything I need is pre-installed, ok. Let's see if I understand! when installing mainsailos, don't I need to run these lines in the terminal?

meteyou, 2023-06-13:
All packages should be installed and ready to use.

docemiel, 2023-06-13:
ok, thanks, i am going to try

meteyou, 2023-06-14:
@docemiel please, write a short feedback after your test.

closed 2023-06-14 by meteyou