mainsail-crew/MainsailOS #239: Feature Request: Enable Klipper Host MCU by default

slashbeast, 2023-08-25:
### Is your feature request related to a problem? Please describe

No.

### Describe the solution you'd like

MainsailOS provides a fantastic out of the box experience. The very only thing that I am missing in all of it is preinstalled by default the 'klipper host mcu' that is required for input shaping calibration. User needs to update klipper config to host, then build and install it and also add systemd service manually to make it work. I think it would make a lot of sense to have it preconfigured along with rest of the software suit .

meteyou, 2023-08-26:
thank you for your inquiry. we have already considered including the host mcu several times, but this has several disadvantages.

first not everyone needs the host MCU, because you can also connect input shaper tests with toolhead boards or with a bluepill/rp2040 via USB.
furthermore it could happen that the host MCU is outdated if you only flashed mainsailOS and then updated Klipper. we don't release a new mainsailOS version often enough.

slashbeast, 2023-08-26:
Those are fair arguments. Thanks.

closed 2023-08-26 by slashbeast