mainsail-crew/MainsailOS #245: Fail to build because of klipper.service "klippy" alias which might be unnecessary.

meganukebmp, 2023-10-20:
### What happened

During build it fails to enable the klipper service as the klippy alias seems to already be taken apparently. I looked around and there doesn't actually seem to be anything using this alias. Removing the alias builds successfully and klipper (and klippy) start correctly.

https://github.com/mainsail-crew/MainsailOS/blob/develop/src/modules/klipper/filesystem/root/etc/systemd/system/klipper.service#L9

There might be a reason for this to exist potentially, hence why I am reporting it as a bug rather than a PR to remove it, as I'm not confident this should be removed. If I install klipper with KIAUH on the same device, it appears to create a service file without the alias.

### What did you expect to happen

Unrealted

### How to reproduce

Unrealted

meganukebmp, 2023-10-20:
> Well, I tried to look what is the reason for that in your repo and saw you are trying to develop for the Zero3.
> 
> How about some contributions to MainsailOS and possibly https://github.com/mainsail-crew/armbian-builds ?
> 
> Maybe we could enhance the current list if builds are valid and well tested? I dont own this SBC but as long you do and do all test, there might be possibilities...
> 
> @meteyou pinging you for reason :P

While I'd love to have this board officially supported this might be stunned by the fact that there is no official armbian support too. There are a few differences between this and the Zero 2 that are currently not mainlined, though I do expect them to be within the next 6 months or so. When that happens it's likely we'll see official armbian too. Right now I'm building it through the official build process they provide here https://github.com/orangepi-xunlong/orangepi-build which happens to be some sort of fork of armbian so it works out in the end though it doesn't make me particularly happy. There's also the fact that the 1.5GB version has a quirk effectively requiring a whole build just for it, when in reality only a few kbytes differ between the images (that being the bootloader).

KwadFan, 2023-10-21:
Well,

We use in our "armbian" build chain also the orangepi config because we need that workaround for the Zero 2.
The armbian kernel isnt well configured, so that you loose wifi at all with a pure armbian.

Therefore, our buildchain builds a "sunxi" variant maintained by orangepi. See https://github.com/mainsail-crew/armbian-builds/tree/master/configs-orangepi.

So, if there is interesst to extend it, please let us know. I would be happy to see contribution like that.

meganukebmp, 2023-10-21:
> Well,
>
> We use in our "armbian" build chain also the orangepi config because we need that workaround for the Zero 2. The armbian kernel isnt well configured, so that you loose wifi at all with a pure [...]

Sure, any explanation on how your build system works? For example, both u-boot and kernel must be pulled manually for the orangepi build system, yet I wasn't able to find that in the workflows.

meteyou, 2023-10-22:
orangepi has a config for it in the "next" branch (all new boards are only in this branch): https://github.com/orangepi-xunlong/orangepi-build/blob/next/external/config/boards/orangepizero3.conf

i added it to our Armbian workflow: https://github.com/mainsail-crew/armbian-builds/pull/37 and here you can test the build: https://github.com/mainsail-crew/armbian-builds/actions/runs/6603359688

Pls test the bullseye and bookworm image and give me a short feedback. if they work, i will merge it and they are available in the releases.

meganukebmp, 2023-10-22:
> orangepi has a config for it in the "next" branch (all new boards are only in this branch): https://github.com/orangepi-xunlong/orangepi-build/blob/next/external/config/boards/orangepizero3.conf
> 
> i added it to our Armbian workflow: [mainsail-crew/armbian-builds#37](https://github.com/mainsail-crew/armbian-builds/pull/37) and here you can test the build: https://github.com/mainsail-crew/armbian-builds/actions/runs/6603359688
> 
> Pls test the bullseye and bookworm image and give me a short feedback. if they work, i will merge it and they are available in the releases.

Are these for the 1/2/4G version or the 1.5G. I only own the 1.5G version and it needs separate builds due to different u-boot.

meteyou, 2023-10-23:
There is no hint or something about different versions. Have you tested the image yet?

meganukebmp, 2023-10-23:
> There is no hint or something about different versions. Have you tested the image yet?

Please see the following:

https://github.com/orangepi-xunlong/orangepi-build/blob/next/scripts/main.sh#L288

https://github.com/orangepi-xunlong/orangepi-build/blob/next/scripts/compilation.sh#L229

https://github.com/orangepi-xunlong/u-boot-orangepi/blob/v2021.07-sunxi/arch/arm/mach-sunxi/dram_sun50i_h616.c#L1313

https://github.com/orangepi-xunlong/u-boot-orangepi/blob/v2021.07-sunxi/arch/arm/mach-sunxi/dram_sun50i_h616.c#L1294

The 1.5GB version carries a different u-boot. Builds for 1/2/4GB fail to boot on the 1.5GB version. I don't currently have the board on hand but will test when I do soon. But please be aware of the differences.

KwadFan, 2023-10-23:
Even a fix is merged already, lets keep this Issue open for discussion about Zero3 Images.

meteyou, 2023-10-24:
Thx for the feedback. I will double-check, how to build this image.

meteyou, 2023-10-24:
ok. i double-checked the code and the option MEM_TYPE should work here. i cannot use hit option in the CUI, but i tested a build with it. it runs without any issue until now. pls test it, when it is finished: https://github.com/mainsail-crew/armbian-builds/actions/runs/6631552667?pr=37

meganukebmp, 2023-10-26:
> ok. i double-checked the code and the option MEM_TYPE should work here. i cannot use hit option in the CUI, but i tested a build with it. it runs without any issue until now. pls test it, when it is finished: https://github.com/mainsail-crew/armbian-builds/actions/runs/6631552667?pr=37

The 1500mb images both boot successfully and connect to the internet.

meteyou, 2023-10-26:
thx for the feedback. then i would merge it to master, that you can use the release images. do you have another question to this topic?

meganukebmp, 2023-10-26:
> thx for the feedback. then i would merge it to master, that you can use the release images. do you have another question to this topic?

Nope, thanks! Glad to have this board supported :)

meteyou, 2023-10-26:
do you want to add support to mainsailOS? this PR will only create a weekly armbian build. nothing more...

meganukebmp, 2023-10-26:
> do you want to add support to mainsailOS? this PR will only create a weekly armbian build. nothing more...

Support for this would be identical to the orangepi zero2 config. We just need 2 configs for the 1500mb and others builds.

KwadFan, 2023-10-28:
> > do you want to add support to mainsailOS? this PR will only create a weekly armbian build. nothing more...
> 
> Support for this would be identical to the orangepi zero2 config. We just need 2 configs for the 1500mb and others builds.

If iz is that easy, will do changes during this week.

Does the overlay, to use SPI on GPIOS work the same as Opi Zero2 ?

meganukebmp, 2023-10-28:
> If iz is that easy, will do changes during this week.
>
> Does the overlay, to use SPI on GPIOS work the same as Opi Zero2 ?

/usr/local/bin/network-configurator doesn't seem to exist on my device. However orangepi firstrun sets up network anyways. Idk how that works.

Here are the optional overlays present:
```bash
orangepi@orangepizero3:/boot/dtb/allwinner/overlay$ ls -lah
total 128K
drwxr-xr-x 2 root root 4.0K Oct 26 18:07 .
drwxr-xr-x 3 root root 4.0K Oct 26 18:07 ..
-rw-r--r-- 1 root root  321 Oct 26 18:03 sun50i-h616-disable-leds.dtbo
-rw-r--r-- 1 root root  228 Oct 26 18:03 sun50i-h616-disable-uart0.dtbo
-rwxr-xr-x 1 root root 4.2K Oct 26 18:03 sun50i-h616-fixup.scr
-rw-r--r-- 1 root root  222 Oct 26 18:03 sun50i-h616-gpu.dtbo
-rw-r--r-- 1 root root  339 Oct 26 18:03 sun50i-h616-ir.dtbo
-rw-r--r-- 1 root root  614 Oct 26 18:03 sun50i-h616-ph-i2c1.dtbo
-rw-r--r-- 1 root root  452 Oct 26 18:03 sun50i-h616-ph-i2c2.dtbo
-rw-r--r-- 1 root root  452 Oct 26 18:03 sun50i-h616-ph-i2c3.dtbo
-rw-r--r-- 1 root root  452 Oct 26 18:03 sun50i-h616-ph-i2c4.dtbo
-rw-r--r-- 1 root root  222 Oct 26 18:03 sun50i-h616-ph-pwm12.dtbo
-rw-r--r-- 1 root root  344 Oct 26 18:03 sun50i-h616-ph-pwm34.dtbo
-rw-r--r-- 1 root root  462 Oct 26 18:03 sun50i-h616-ph-uart2.dtbo
-rw-r--r-- 1 root root  462 Oct 26 18:03 sun50i-h616-ph-uart5.dtbo
-rw-r--r-- 1 root root  712 Oct 26 18:03 sun50i-h616-pi-i2c0.dtbo
-rw-r--r-- 1 root root  344 Oct 26 18:03 sun50i-h616-pi-i2c1.dtbo
-rw-r--r-- 1 root root  564 Oct 26 18:03 sun50i-h616-pi-i2c2.dtbo
-rw-r--r-- 1 root root  439 Oct 26 18:03 sun50i-h616-pi-pwm1.dtbo
-rw-r--r-- 1 root root  439 Oct 26 18:03 sun50i-h616-pi-pwm2.dtbo
-rw-r--r-- 1 root root  439 Oct 26 18:03 sun50i-h616-pi-pwm3.dtbo
-rw-r--r-- 1 root root  439 Oct 26 18:03 sun50i-h616-pi-pwm4.dtbo
-rw-r--r-- 1 root root  462 Oct 26 18:03 sun50i-h616-pi-uart2.dtbo
-rw-r--r-- 1 root root  462 Oct 26 18:03 sun50i-h616-pi-uart3.dtbo
-rw-r--r-- 1 root root  462 Oct 26 18:03 sun50i-h616-pi-uart4.dtbo
-rw-r--r-- 1 root root  619 Oct 26 18:03 sun50i-h616-spi0-spidev.dtbo
-rw-r--r-- 1 root root  871 Oct 26 18:03 sun50i-h616-spi1-cs0-cs1-spidev.dtbo
-rw-r--r-- 1 root root  706 Oct 26 18:03 sun50i-h616-spi1-cs0-spidev.dtbo
-rw-r--r-- 1 root root  706 Oct 26 18:03 sun50i-h616-spi1-cs1-spidev.dtbo
-rw-r--r-- 1 root root  608 Oct 26 18:03 sun50i-h616-usb0-host.dtbo
-rw-r--r-- 1 root root  285 Oct 26 18:03 sun50i-h616-zero2w-disable-led.dtbo
```

from `orangepiEnv.txt`: `overlay_prefix=sun50i-h616`. So Enabling overlays would be very similar. In this case they seem to have an additional port prefix. So instead of just `uart5` it will be `ph-uart5` or instead

meganukebmp, 2023-10-28:
I just tested.
`overlays=ph-uart5` as an example, applies correctly.

Uboot log:
```
Applying kernel provided DT overlay sun50i-h616-ph-uart5.dtbo
1055 bytes read in 2 ms (514.6 KiB/s)
```

meganukebmp, 2023-10-28:
Sorry, I didn't read source correctly. `network-configurator` originates in your repo. `/usr/lib/orangepi/orangepi-firstrun-config` is present on the zero3 same as the zero2.

meganukebmp, 2023-10-28:
> I just tested. `overlays=ph-uart5` as an example, applies correctly.
> 
> Uboot log:
> 
> ```
> Applying kernel provided DT overlay sun50i-h616-ph-uart5.dtbo
> 1055 bytes read in 2 ms (514.6 KiB/s)
> ```

This seems to be the only actual difference. Here's a reference of the zero3 pin alt functions
[image]

KwadFan, 2023-10-28:
> Sorry, I didn't read source correctly. `network-configurator` originates in your repo. `/usr/lib/orangepi/orangepi-firstrun-config` is present on the zero3 same as the zero2.

Exactly this one is deactivated to apply Network configuration as described in https://docs-os.mainsail.xyz/faq/change-wifi-credentials

KwadFan, 2023-10-28:
For reference:

https://github.com/mainsail-crew/MainsailOS/blob/6c95a7bb4075d7cac01fa4c63088da5189d26714/src/modules/orangepi_net/start_chroot_script#L47

https://github.com/mainsail-crew/MainsailOS/blob/6c95a7bb4075d7cac01fa4c63088da5189d26714/src/modules/orangepi_net/start_chroot_script#L38

KwadFan, 2023-10-28:
Btw. would you mind joining our discord ? discord.gg/mainsail.

We could open a thread there for realtime discussion if you are interessted in that.

PR #246 (2023-10-23): fix: removing 'klippy' alias
Seems the used alias, which is uneccessary at all, leads to a failing build in some configurations.

This should fix #245
commit c8521b2c (2023-10-23): fix: removing 'klippy' alias (#246)

Seems the used alias, which is uneccessary at all, leads to
a failing build in some configurations.

This should fix #245
closed 2025-06-07 by meteyou