mainsail-crew/MainsailOS #251: change /etc/lsb-release and /etc/os-release to make it more visible to other programs that one is using mainsailOS, e.g. neofetch

Online122228, 2023-10-29:
### Is your feature request related to a problem? Please describe

neofetch only shows raspbian logo and i do not like that

### Describe the solution you'd like

as seen in headline

meteyou, 2023-10-29:
pls use a short headline and use the description to describe your issue as good as possible.

PS: kwadfan will answer here with a real feedback...

KwadFan, 2023-10-31:
We can not change the os-release file. This would require we alter all packages from Raspberry Foundation/armbian/orange-pi to not disrupt possible workflows that relie on the original os-release file.

If you want to show it in neofetch, you should consider to seperate that and also looking for `/etc/mainsailos-release`file ...

Other than that I wouldnt describe it as an issue if a thirdparty tool (even I love neofetch too) related to the OS.

Therefore, closing that, sorry mate but out of our scope...

closed 2023-10-31 by KwadFan

Online122228, 2023-10-31:
I created a fork of neofetch and after changing the /etc/os-release back, it still worked and showed the mainsailOS ascii-logo i created, so i read a bit about it and found out that neofetch actually looks for files like the /etc/mainsailos-release. So if one wants neofetch with mainsailOS ascii art(small version included) you can find it at github.com/online122228/neofetch

KwadFan, 2023-11-01:
> I created a fork of neofetch and after changing the /etc/os-release back, it still worked and showed the mainsailOS ascii-logo i created, so i read a bit about it and found out that neofetch [...]

You really should consider to PR to dylana's repo ...
Thanks for your work

Online122228, 2023-11-01:
Already did and created a .deb file