mainsail-crew/MainsailOS #262: crowsnest.conf not updated for non-pi user

john3300, 2023-12-19:
### What happened

I installed Mainsail OS using the Raspberry Pi Imager, but I changed the user to something other than the default 'pi'. I noticed that the crowsnest.conf still references the default pi user in:
`log_path: /home/pi/printer_data/logs/crowsnest.log`.

### What did you expect to happen

I expected the `/home/pi` path to be updated for the non-default user.

### How to reproduce

Install Mainsail OS using the Raspberry Pi Imager and a non-default user.

KwadFan, 2024-01-02:
Have to do some tests, but I rename my user usually and it never happend to me.
Possible you didnt wait long enough to let "postrename" script to finish.

KwadFan, 2024-01-02:
Ok, I think I forgot something...
#235

KwadFan, 2024-02-21:
This could be closed, already fixed, merged and released...

closed 2024-02-21 by KwadFan