mainsail-crew/MainsailOS #263: toolchanger and mmu frontend support with spoolman

Marshalldog, 2024-01-01:
### Is your feature request related to a problem? Please describe

Hey. Currently building a tool changer and I'm using spoolman to track filament usage. I asked if spoolman supports tracking multiple filaments (for multiple toolheads) and the dev confirmed its supported and just needs to be added on the mainsail frontend: https://github.com/Donkie/Spoolman/issues/239

### Describe the solution you'd like

here is an idea for the UI.
So each extruder can be assigned a different filament
[image]

meteyou, 2024-01-01:
Thx for your request. i have to build a test rig to implement this feature.

meteyou, 2024-01-02:
@Marshalldog I have only now seen that you have made the FR in the wrong repo. this is only for the SBC image.

closed 2024-01-02 by meteyou