mainsail-crew/MainsailOS #282: Regularly stuck at "Initializing..." using web UI

esb7, 2024-01-27:
### What happened

I regularly get stuck at "Initializing..." when accessing mainsail from a web browser on my Mac. Sometimes it works and other times it fails and will continue to fail even through reboots of the rPi hosting mainsail. It would occur on different web browsers as well.

### What did you expect to happen

Web UI should be consistent and responsive, not hung.

### How to reproduce

Use mainsailos.local to access the printer web UI.

### Additional information

The issue seems caused by "unauthorized" errors that show up in the moonraker log:

> ```2024-01-20 18:50:33,939 [websockets.py:on_close()] - Websocket Closed: ID: 2822839248 Close Code: 1001, Close Reason: None, Pong Time Elapsed: 8.53 2024-01-20 18:50:34,137 [websockets.py:prepare()] - Websocket Failed Authentication: HTTP 401: Unauthorized (Unauthorized) 2024-01-20 18:50:34,140 [app.py:log_request()] - 101 GET /websocket (fd70:616c:c5db:914e:18f6:e20b:5620:9152) [No User] 23.93ms ```

I noticed that the failed authentications all had a reported ipv6 address and on further investigation were all in a range outside of the allowed range in the `trusted_clients:` section in `moonraker.conf`. The primary / default trusted ipv6 range is `FE80::/10`. However that appear to exclude other local-only address ranges (including the FD70 my device had). I believe the default should be: `FC00::/7`.

For anybody that hits this, you can get past it by adding the following to the trusted clients: section of the moonraker.conf file:

```
[authorization]
...
trusted_clients:
...
    FC00::/7        <------------- add this
...```

meteyou, 2024-01-27:
FE80::/10 is reservated for local ip addresses in the ipv6 network. if you network use a different network, you have to add it. you can read more informations about FE80::/10 for example here: https://en.wikipedia.org/wiki/IPv6#Link-local_address

esb7, 2024-01-27:
From what I can tell, there is a larger "Unique Local Address" range defined for IPv6 that is subtly different than link local, but is equivalent to the non-publicly routable 10.0.0.0/8, 172.16.0.0/12 and 192.168.0.0/16 ranges in IPv4:

https://en.wikipedia.org/wiki/Unique_local_address

My local network (using UniFi network gear) appears to be set up by default to allocate IPv6 addresses in the larger FC00::/7 range (outside of the link-local range).  As a result, I get the above "Initializing..." hang / "unauthorized" errors.

It seems like it should be universally safe to include the larger "Unique Local" range by default.

meteyou, 2024-01-27:
Ah ok. I didn't know that. I will double check it in the team and maybe we will add it.

reneas, 2024-06-27:
I have a similar problem but without the `unauthorized` part. For me `klippy.log` reads:

```
[websockets.py:on_close()] - Websocket Closed: ID: 548082250224 Close Code: None, Close Reason: None, Pong Time Elapsed: 3.04
```
and i have a `mainsail-error.log`:
```
2024/06/27 21:14:01 [error] 721#721: *24 recv() failed (104: Connection reset by peer) while proxying upgraded connection, client: 192.168.10.33, server: _, request: "GET /websocket HTTP/1.1", upstream: "http://127.0.0.1:7125/websocket", host: "192.168.10.23"
2024/06/27 21:14:19 [error] 722#722: *1 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.10.33, server: _, request: "GET /websocket HTTP/1.1", upstream: "http://127.0.0.1:7125/websocket", host: "192.168.10.23"
2024/06/27 21:14:20 [error] 722#722: *3 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.10.33, server: _, request: "GET /websocket HTTP/1.1", upstream: "http://127.0.0.1:7125/websocket", host: "192.168.10.23"
2024/06/27 21:14:22 [error] 722#722: *5 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.10.33, server: _, request: "GET /websocket HTTP/1.1", upstream: "http://127.0.0.1:7125/websocket", host: "192.168.10.23"
2024/06/27 21:14:23 [error] 722#722: *7 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.10.33, server: _, request: "GET /websocket HTTP/1.1", upstream: "http://127.0.0.1:7125/websocket", host: "192.168.10.23"
2024/06/27 21:14:24 [error] 722#722: *9 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.10.33, server: _, request: "GET /websocket HTTP/1.1", upstream: "http://127.0.0.1:7125/websocket", host: "192.168.10.23"
2024/06/27 21:14:24 [error] 722#722: *11 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.10.33, server: _, request: "GET /websocket HTTP/1.1", upstream: "http://127.0.0.1:7125/websocket", host: "192.168.10.23"
2024/06/27 21:14:25 [error] 722#722: *13 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.10.33, server: _, request: "GET /websocket HTTP/1.1", upstream: "http://127.0.0.1:7125/websocket", host: "192.168.10.23"
2024/06/27 21:15:10 [error] 722#722: *1 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.10.33, server: _, request: "GET /websocket HTTP/1.1", upstream: "http://127.0.0.1:7125/websocket", host: "192.168.10.23"
2024/06/27 21:15:11 [error] 722#722: *3 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.10.33, server: _, request: "GET /websocket HTTP/1.1", upstream: "http://127.0.0.1:7125/websocket", host: "192.168.10.23"
2024/06/27 21:15:12 [error] 722#722: *5 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.10.33, server: _, request: "GET /websocket HTTP/1.1", upstream: "http://127.0.0.1:7125/websocket", host: "192.168.10.23"
2024/06/27 21:15:13 [error] 722#722: *7 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.10.33, server: _, request: "GET /websocket HTTP/1.1", upstream: "http://127.0.0.1:7125/websocket", host: "192.168.10.23"
2024/06/27 21:15:14 [error] 722#722: *9 connect() failed (111: Connection refused) while connecting to upstream, client: 192.168.10.33, server: _, request: "GET /websocket HTTP/1.1", upstream: "http://127.0.0.1:7125/websocket", host: "192.168.10.23"

```

adding the lines suggested by @esb7 seems to have fixed it (for now)

ChrisRenfrow, 2024-08-12:
If anyone is coming here because they are trying to use Tailscale to remotely connect to MainsailOS from within their tailnet, you will need to ensure that the ips of the connecting clients fall within the ranges defined in moonraker.conf's [authorization.trusted_clients](https://moonraker.readthedocs.io/en/latest/configuration/#authorization). Whether you use a range or verbatim ip address(es) is up to you (I personally just added the ip addresses of the devices I wanted to allow access).

Perhaps this warrants an error message that makes it more clear that Moonraker says the client is unauthorized?

julianCast, 2024-09-03:
I was having same issue for different reason, I'll post it here in case it helps others:

I didn't have a non default ipv6 but this:
```
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.152  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::454c:3df7:2e55:76dt  prefixlen 64  scopeid 0x20<link>
        inet6 2a0c:6a80:9903:4500:6fbf:547c:d0ee:32d3  prefixlen 64  scopeid 0x0<global>
        ether b8:17:eb:ad:ef:61  txqueuelen 1000  (Ethernet)
        RX packets 1227  bytes 197698 (193.0 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 579  bytes 92243 (90.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

So I don't know if it makes sense but  it started working consistently after adding this to the `printer_data/config/moonraker.conf`:
```
FE00::/7     --> just in case
192.168.1.0/24
2a0c:6a80:9903:4500::/64
```

julianCast, 2024-09-04:
> I having same issue for different reason:
> 
> I didn't have a non default ipv6 but this:
> 
> ```
> wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
>         inet 192.168.1.152  netmask 255.255.255.0  broadcast 192.168.1.255
>         inet6 fe80::454c:3df7:2e55:76dt  prefixlen 64  scopeid 0x20<link>

Updated since I'm still having the issue

junglestem, 2024-09-24:
> If anyone is coming here because they are trying to use Tailscale to remotely connect

Solved my problem...thanks!  Loving the Tailscale remote access.

closed 2025-08-15 by meteyou