mainsail-crew/MainsailOS #287: Raspberry Pi 5 Optimization

JorgeSDF, 2024-02-29:
### Is your feature request related to a problem? Please describe

I saw already various users wanting to use the Rpi5 for mainsail UI and Klipper instalation. I own a Rpi5 and im running Mainsail on it for my Vzbot.

### Describe the solution you'd like

It would be nice to have a Mainsail image on Raspberry Pi Imager that is optimize for Rpi5 to have better camera performance and utilize the power.

### Describe alternatives you've considered

Or only make an optimize version for Rpi5 on the imager

meteyou, 2024-02-29:
Thanks for your request. We are working on it! 

Unfortunately, the webcam area will be tricky with the rpi5, as it has no HW encoder. This means that only software rendering is possible, and currently, only ustreamer with MJPEG will work on it. WebRTC is presently not possible with the rpi5.

KwadFan, 2024-02-29:
Mentioning https://github.com/mainsail-crew/MainsailOS/pull/281

meteyou, 2025-06-07:
Fixed with version v2.0.0 -> https://github.com/mainsail-crew/MainsailOS/releases/tag/2.0.0

closed 2025-06-07 by meteyou