mainsail-crew/MainsailOS #295: Add ability to use Pi 5

ankonaskiff17, 2024-05-15:
### Is your feature request related to a problem? Please describe

Please add Pi 5 as appropriate for last line. Not sure if lacking Pi 5 would cause installation to fail or not but currently only options Pi 3 and Pi 4

config/raspberry/rpi64

#!/usr/bin/env bash
# Shebang for better file detection

# Declare Variables before exporting.
# See https://www.shellcheck.net/wiki/SC2155

BASE_ARCH="arm64"

# export variables
export BASE_ARCH

### JSON sniplet Setup
### NOTE: Please see all config files for setup variables!!!
# shellcheck disable=SC2034
JSON_SUPPORTED_SBC="pi3-64bit pi4-64bit"

### Describe the solution you'd like

Add Pi 5 to Raspberry Pi SBC's allowed

meteyou, 2024-05-15:
raspberry pi5 needs bookworm. right now, mainsail only use debian bookworm. so when we add pi5, it will not boot on a pi5.

keefo, 2024-06-01:
Backup on this request.

This doc says the debian bookworm is ready for pi5.
https://qengineering.eu/bookworm.html

meteyou, 2024-06-01:
@keefo sry... I miswritten the post before. MainsailOS use bullseye.

andresousadotpt, 2025-04-26:
Any updates regarding enabling mainsailos for Pi 5?

meteyou, 2025-04-26:
@andresousadotpt https://github.com/mainsail-crew/MainsailOS/pull/314

if you want to be up-to date, you can also follow me per patreon to read the news monthly.

ankonaskiff17, 2025-04-26:
Good luck. 

I bought a Bambu Lab printer because Mainsail is years out of date and I got tired of waiting for updates .

I suspect if one can see a trend of Mainsail OS downloads per month over the last two years, the number of downloads are declining 

 

 

 

Wayne Wells

email: ***@***.***

Cell: (803) 553-6091

 

10682 Two Notch Rd.

Apt. 7107

Elgin, SC 29045

meteyou, 2025-04-27:
@ankonaskiff17 I understand it can be frustrating when an open-source project doesn’t move as quickly as commercial products. However, it’s important to remember that Mainsail is a hobby project developed by volunteers in their spare time—unlike companies like Bambu Lab, which have large teams and significant financial resources.

Mainsail is definitely not “out of date.” We release updates regularly (the last Mainsail update was just 4 days ago!) and are always working on improvements. Of course, with more support—whether it’s contributions, feedback, or even financial backing—we could achieve even more. If you care about the project, we’d really appreciate any form of involvement!

Constructive criticism and engagement help us move forward, but just complaining doesn’t. We invite you to get involved or support the project so we can continue to develop Mainsail together.

Regarding MainsailOS: Progress has been slow mainly because our workflow framework (CustomPiOS) was stuck, and we waited a long time for our pull requests to be merged to support the new network stack from RPi OS Bookworm. Since this didn’t happen, we had to start over from scratch and rebuild MainsailOS with a different framework over the past two months. I personally spent all my free time refactoring it.

Additionally, we don’t see much benefit in using an RPi 5 for 3D printers—there are actually some downsides, such as the removal of hardware encoders. This also reduced our motivation to support it. However, you can still use a generic RPi OS Lite and install Klipper, Moonraker, and Mainsail via KIAUH. The end result is exactly the same as using MainsailOS. MainsailOS is simply the “faster way” to get started with a Klipper printer.

Lastly, I’d be genuinely interested to hear more details from you about where you feel Mainsail is “out of date,” with specific examples if possible. This kind of feedback would be very helpful for us to improve the project. Thanks again for your input!

meteyou, 2025-06-07:
fixed with https://github.com/mainsail-crew/MainsailOS/releases/tag/2.0.0

closed 2025-06-07 by meteyou