mainsail-crew/MainsailOS #300: Display UI error(s) when interface is unable to load

ChrisRenfrow, 2024-08-12:
### Is your feature request related to a problem? Please describe

There are situations in which the interface can be left hanging on "Initializing" with no indication of any errors with the process outside of the console. Additionally, these console errors can be difficult for users to understand.

[image]

See: https://github.com/mainsail-crew/MainsailOS/issues/282

### Describe the solution you'd like

I think at a minimum there should be some indicator in the UI that errors have occurred along with some suggested actions to troubleshoot or seek support. Ideally the error, if known, should be explained in terms that are more easily understood with suggestions for troubleshooting.

e.g. 
> **Error:** Moonraker - Client {ip} untrusted! Check your `moonraker.conf`'s [authorization module](https://moonraker.readthedocs.io/en/latest/configuration/#authorization), specifically `trusted_clients`.

### Describe alternatives you've considered

Alternatively, or in addition to, the console errors themselves could offer a bit more information to help users troubleshoot/identify the issue.

meteyou, 2024-08-12:
@ChrisRenfrow pls don't mixup Mainsail with MainsailOS. MainsailOS is just a rpi image with preinstalled Klipper, Moonraker and Mainsail. Mainsail is the Webinterface.

closed 2024-08-12 by meteyou

ChrisRenfrow, 2024-08-12:
Ah, my apologies. Thanks for the correction!