mainsail-crew/MainsailOS #345: MainsailOS 2.1.0 fails to flash device is DFU mode

Vitalii3D-xyz, 2025-08-19:
### What happened

# Setup*
mcu - Manta M8P V2.0
controller - CM5

_*I also have another M8P V2.0 and M8P V1.0 and another CM5. I can reproduce this issue on any combinations of those MCU + CM5 module_
# What Happened
Flashing newest klipper on MCU in DFU mode fails with next error:
```
dfu-util: error get_status: LIBUSB_ERROR_TIMEOUT
```

### What did you expect to happen

MCU flashes correctly, no errors. 
I have verified everything works as expected by moving back to MainsailOS V2.0.0

### How to reproduce

1. Install MainsailOS 2.1.0 on CM5 eMMC
2. Put Manta M8P in DFU mode, confirm by running `lsusb`
3. Compile klipper for M8P, all configuration is taken from BTT manual
4. Flash compiled klipper on MCU

### Additional information

```
$ uname -a
Linux raspberryvoron 6.12.34+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.34-1+rpt1~bookworm (2025-06-26) aarch64 GNU/Linux
```
```
$ lsusb
Bus 001 Device 004: ID 0483:df11 STMicroelectronics STM Device in DFU Mode
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
```
$ make flash FLASH_DEVICE=0483:df11
  Flashing out/klipper.bin to 0483:df11
sudo dfu-util -d ,0483:df11 -R -a 0 -s 0x8020000:leave -D out/klipper.bin

dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
Opening DFU capable USB device...
Device ID 0483:df11
Device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
dfu-util: error get_status: LIBUSB_ERROR_TIMEOUT

Failed to flash to 0483:df11: Error running dfu-util

If the device is already in bootloader mode it can be flashed with the
following command:
  make flash FLASH_DEVICE=0483:df11
  OR
  make flash FLASH_DEVICE=1209:beba

If attempting to flash via 3.3V serial, then use:
  make serialflash FLASH_DEVICE=0483:df11

make: *** [src/stm32/Makefile:107: flash] Error 255
```

mryel00, 2025-08-19:
> I have verified everything works as expected by moving back to MainsailOS V2.0.0

Just to confirm: You updated everything on MainsailOS V2.0.0, before you tried to install everything?
Please make sure that everything is up to date even on the MainsailOS V2.0.0 image with 
```sh
sudo apt update
sudo apt upgrade
cd klipper
git pull
```
Report back if it is working on MainsailOS V2.0.0 after running these commands.

Vitalii3D-xyz, 2025-08-19:
So I have installed V2.1.0 again and now I can't flash MCU anymore. Happy to test any idea you have as I have spare Manta + CM5. This is my test setup:

[image]

mryel00, 2025-08-19:
Can you test the latest PiOS lite too?

Vitalii3D-xyz, 2025-08-19:
> Can you test the latest PiOS lite too?

Do you mean install it, then install klipper manually and then try to flash MCU? I can try tonight (PDT Zone). May take me some time as last time I did that many years ago

meteyou, 2025-08-19:
just flashing raspberry pi os lite + upgrade it + run the DFU commands should be enough. i think you dont need to install klipper for this test.

Vitalii3D-xyz, 2025-08-19:
Okay, so latest Pi OS Lite works. I have used KIAUH to install all dependencies, then compiled klipper for MCU and flashed it. But Pi OS Lite is behind Mainsail OS V2.1.0, it is based Debian build from **2025-04-30**, while Mainsail OS V2.1.0 is based on build from **2025-06-26**
```
$ uname -a
Linux cm5test 6.12.25+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux
```

Vitalii3D-xyz, 2025-08-20:
> Did you run apt upgrade and reboot?

In what configuration?

meteyou, 2025-08-20:
On the normal Raspi os. This should upgrade it to the same state than MainsailOS.

Vitalii3D-xyz, 2025-08-20:
> On the normal Raspi os. This should upgrade it to the same state than MainsailOS.

I didn't run update in this case, but I have tried that with MainsailOS V2.0.0
Would you like me to install Pi OS Lite, run update, upgrade then reboot and then try dfu again?
Is it ok if I will try that on existing Pi OS Lite that I did't upgrade yet?

Vitalii3D-xyz, 2025-08-20:
An just in case, I also had an issue with NH36 V1.4 board, this is log from attaching it:
```
$ dmesg -HW
[Aug19 00:49] usb 1-1.1: new full-speed USB device number 7 using dwc2
[  +0.204207] usb 1-1.1: not running at top speed; connect to a high speed hub
[  +0.000500] usb 1-1.1: New USB device found, idVendor=1a86, idProduct=8091, bcdDevice=12.10
[  +0.000005] usb 1-1.1: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[  +0.000003] usb 1-1.1: Product: USB HUB
[  +0.000254] hub 1-1.1:1.0: USB hub found
[  +0.001890] hub 1-1.1:1.0: 4 ports detected
[  +0.000073] hub 1-1.1:1.0: config failed, can't get hub status (err -5)
```
I can't reproduce this issue after downgrading to MainsailOS V2.1.0

Vitalii3D-xyz, 2025-08-20:
@meteyou after running:
```
sudo apt-get update
sudo apt-get upgrade
sudo shutdown - r now
```

Core is still form 2025-04-30:
```
ux raspberryvoron 6.12.25+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux
```

meteyou, 2025-08-20:
it would be "funny" if RPI remove this package after we build MainsailOS v2.1.0 from the repo.

Our MainsailOS workflow just pull the latest RPI OS lite image, run apt update & apt upgrade and then install all components. This was the exact run of the release build: https://github.com/mainsail-crew/MainsailOS/actions/runs/16988420772/job/48162220868#step:13:262

meteyou, 2025-09-07:
A short update here. I go very deep with Esoterical and AtomicVectorPlotter on the Voron discord and we found out, that this issue happens with this Raspberry Pi Kernel:
```
Linux raspberrypi 6.12.34+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.12.34-1+rpt1~bookworm (2025-06-26) aarch64 GNU/Linux
```

This issue was already detected on the Raspberry Kernel repo and is already fixed: https://github.com/raspberrypi/linux/pull/6936

Here also a discussion about this issue and a comment how to install this test Kernel: https://github.com/raspberrypi/linux/issues/6941#issuecomment-3041164256

I tested it and it works fine with this fix. Now we have to wait for the next Raspberry Pi Kernel. This should happen in next week(s). I will release a new MainsailOS after this, because the MainsailOS workflow update the RpiOS image during the build process.

closed 2025-10-04 by meteyou